gazebosim/gazebo-classic #3316: Disable linux-build CI job?

traversaro, 2023-04-15:
Since a few months, the `linux-build` GitHub Action CI job has been failing with error:
~~~
Fetched 6294 kB in 1s (4851 kB/s)
Reading package lists...
W: http://packages.osrfoundation.org/gazebo/ubuntu-stable/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
--2023-04-15 13:43:56--  https://raw.githubusercontent.com/ignition-tooling/release-tools/master/jenkins-scripts/lib/dependencies_archive.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.108.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 27772 (27K) [text/plain]
Saving to: ‘/tmp/dependencies.sh’

     0K .......... .......... .......                         100% 30.9M=0.001s

2023-04-15 13:43:56 (30.9 MB/s) - ‘/tmp/dependencies.sh’ saved [27772/27772]

Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package libogre-2.1-dev
E: Couldn't find any package by glob 'libogre-2.1-dev'
E: Couldn't find any package by regex 'libogre-2.1-dev'
Error: Process completed with exit code 123.
~~~

To reduce the CI noise, could it make sense to remove it? @scpeters @j-rivero

j-rivero, 2023-04-17:
> To reduce the CI noise, could it make sense to remove it? @scpeters @j-rivero

No problem by my side.

scpeters, 2023-04-18:
> > To reduce the CI noise, could it make sense to remove it? @scpeters @j-rivero
> 
> No problem by my side.

I agree

traversaro, 2023-04-18:
I just noticed this job was created by @talregev in https://github.com/gazebosim/gazebo-classic/pull/3049 . @talregev it is ok if we disable that CI job that has been broken for a long time? Thanks!

scpeters, 2023-04-23:
I just manually disabled the Linux workflow from the GitHub web U/I based on the instructions from the following page:

* https://docs.github.com/en/actions/managing-workflow-runs/disabling-and-enabling-a-workflow

traversaro, 2023-04-23:
Thanks, I guess we can close this issue then.

talregev, 2023-04-23:
@traversaro @scpeters 
Can you also comments regards the PR I did? 
Thank you.

traversaro, 2023-04-23:
> @traversaro @scpeters 
> Can you also comments regards the PR I did? 
> Thank you.

Sure, I will check it.