gazebosim/gazebo-classic #3343: question: does gazebo-11 not support damping?

sizsJEon, 2023-09-07:
I'm trying to make a robot with some parameters, 

but it seems to be damping is not working properly.

for example,  in sdf file, dynamic's friction is working properly, but damping is not.

    <joint name='JOINT' type='revolute'>
      <pose relative_to='BODY'>0.31804 0.1138 0 0 -0 0</pose>
      <parent>BODY</parent>
      <child>LINK1</child>
      <axis>
        <xyz>1 0 0</xyz>
        <dynamics damping="0.2" friction="0.2"/>
      </axis>

also, this->JOINT->SetDamping(0.2, 0.2 );

is not working too.
 
(I checked it with movement of joint and rosservice command)
rosservice call /gazebo/get_joint_properties "joint_name: 'JOINT'"
type: 0
damping: []
position: [-1.0584862597740239]
rate: [-2.370903274688811]
success: True
status_message: "GetJointProperties: got properties"

any suggestion will be helpful.
Regards.

traversaro, 2023-09-07:
Which physics engine are you using? The default one, i.e. ode?

sizsJEon, 2023-09-07:
> 

yes. I use default one.

    <physics name="default_physics" default="0" type="ode">

sizsJEon, 2023-09-07:
> Which physics engine are you using? The default one, i.e. ode?

yes. I use default one.

<physics name="default_physics" default="0" type="ode">

scpeters, 2023-09-13:
> ```
> <joint name='JOINT' type='revolute'>
>   <pose relative_to='BODY'>0.31804 0.1138 0 0 -0 0</pose>
>   <parent>BODY</parent>
>   <child>LINK1</child>
>   <axis>
>     <xyz>1 0 0</xyz>
>     <dynamics damping="0.2" friction="0.2"/>
>   </axis>
> ```

this looks like a mix of SDFormat and URDF syntax

Using the [joint_damping_demo.world](https://github.com/gazebosim/gazebo-classic/blob/af7d3278a1cfae54328bfd2d4eba7b1c29ee669c/worlds/joint_damping_demo.world#L324-L336) as an example, I would change your axis tag as follows to use SDFormat syntax:

~~~
  <axis>
    <xyz>1 0 0</xyz>
    <dynamics>
      <damping>0.2</damping>
      <friction>0.2</friction>
    </dynamics>
  </axis>
~~~

sizsJEon, 2023-09-13:
thanks for the update, i'll try it!

closed 2023-09-13 by sizsJEon