gazebosim/gazebo-classic #3357: Gazebo Classec (gazebo11) can't be installed on Ubuntu 22.04 Jammy with ROS2 Humble.

MuhammadQaisarAli, 2023-11-02:
**Ubuntu 22.04 Jammy** 
**ROS2 Humble** installed from binary via (https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html)

Trying to install Gazebo11 classic but unable. 
I am trying to install using this method ( https://classic.gazebosim.org/tutorials?tut=install_ubuntu)

I will attache complete terminal data for installation process. Please note that I have tried more than 10 times on different machines, but no fix.

```bash
root@localhost:~# sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
root@localhost:~# cat /etc/apt/sources.list.d/gazebo-stable.list
deb http://packages.osrfoundation.org/gazebo/ubuntu-stable jammy main
root@localhost:~# wget https://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
--2023-11-02 18:14:42--  https://packages.osrfoundation.org/gazebo.key
Resolving packages.osrfoundation.org (packages.osrfoundation.org)... 52.52.171.73
Connecting to packages.osrfoundation.org (packages.osrfoundation.org)|52.52.171.73|:443... connected.
HTTP request sent, awaiting response... Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
200 OK
Length: 1755 (1.7K) [application/octet-stream]
Saving to: ?STDOUT?

-                               100%[====================================================>]   1.71K  --.-KB/s    in 0s      

2023-11-02 18:14:44 (110 MB/s) - written to stdout [1755/1755]

OK
root@localhost:~# sudo apt-get update
Hit:1 http://packages.osrfoundation.org/gazebo/ubuntu-stable jammy InRelease                                                
Hit:2 http://packages.ros.org/ros2/ubuntu jammy InRelease                                                                   
Get:3 http://packages.osrfoundation.org/gazebo/ubuntu-stable jammy/main arm64 Packages [105 kB]
Hit:4 http://ports.ubuntu.com/ubuntu-ports jammy InRelease         
Hit:5 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease
Hit:6 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease
Hit:7 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease                                                         
Fetched 105 kB in 11s (9,641 B/s)                                                                                           
Reading package lists... Done
W: http://packages.osrfoundation.org/gazebo/ubuntu-stable/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
root@localhost:~# sudo apt-get install gazebo11
# For developers that work on top of Gazebo, one extra package
sudo apt-get install libgazebo11-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package gazebo11 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'gazebo11' has no installation candidate
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libgazebo11-dev
root@localhost:~# 
```

Rob-Dawson, 2023-11-06:
Just use 
`sudo apt install gazebo`
This installs Gazebo Classic Version 11. I'm running ROS 2 Humble on Ubuntu 22.04 and have both Gazebo Classic and Gazebo Ignition (Fortress built from source)

MuhammadQaisarAli, 2023-11-07:
> Just use `sudo apt install gazebo` This installs Gazebo Classic Version 11. I'm running ROS 2 Humble on Ubuntu 22.04 and have both Gazebo Classic and Gazebo Ignition (Fortress built from source)

```bash

root@localhost:~# sudo apt install gazebo
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package gazebo is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'gazebo' has no installation candidate

```

roncapat, 2023-11-07:
@Rob-Dawson you may had `gazebo` installed prior to upgrade to 22.04, am I right? It is my understanding that `gazebo-classic` can't be installed anymore from repository on Ubuntu 22.04.

traversaro, 2023-11-07:
Hello everyone! For this kind of question-like content, it may be more indicate to post on http://robotics.stackexchange.com/ . 

For installing Gazebo Classic on Ubuntu 22.04, you need to run `sudo apt install libgazebo-dev`. However, note that this is a package provided by the official Ubuntu repositories, and not by the not by Gazebo mantainers, so the version provided is 11.10.2, and typically no update is typically provided.

Rob-Dawson, 2023-11-07:
> @Rob-Dawson you may had `gazebo` installed prior to upgrade to 22.04, am I right? It is my understanding that `gazebo-classic` can't be installed anymore from repository on Ubuntu 22.04.

No. I have just now installed 22.04 on a virtual machine and installed Gazebo classic using the method above without issue. @traversaro Is correct in that the version is 11.10.2

MuhammadQaisarAli, 2023-11-07:
> Hello everyone! For this kind of question-like content, it may be more indicate to post on http://robotics.stackexchange.com/ .
> 
> For installing Gazebo Classic on Ubuntu 22.04, you need to run `sudo apt install libgazebo-dev`. However, note that this is a package provided by the official Ubuntu repositories, and not by the not by Gazebo mantainers, so the version provided is 11.10.2, and typically no update is typically provided.
> 
> See https://classic.gazebosim.org/tutorials?tut=install_ubuntu for more details.

Thank you very much for your detailed response. 
But if you please look to my problem statement, I already had mentioned and followed the link which you provided.

Further, I ran the command provided by you, and I have the following output.

```bash

root@localhost:~# sudo apt install libgazebo-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libgazebo-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libgazebo-dev' has no installation candidate

```

MuhammadQaisarAli, 2023-11-07:
> @Rob-Dawson you may had `gazebo` installed prior to upgrade to 22.04, am I right? It is my understanding that `gazebo-classic` can't be installed anymore from repository on Ubuntu 22.04.

I totally agree with you.

Few weeks ago, there was no problem in installation of gazebo Classic for me on many machines. But recently it seems a problem and can't be solved, at least by me.

MuhammadQaisarAli, 2023-11-07:
> No. I have just now installed 22.04 on a virtual machine and installed Gazebo classic using the method above without issue. @traversaro Is correct in that the version is 11.10.2

I am not sure, when you installed.

But can you please install it again on virtual machine?
Can you please follow the steps which I followed? 
I means installing Ubuntu 22.04, the ROS2 Humble, and then Gazebo 11 Classic?.

Rob-Dawson, 2023-11-07:
I tried the installation earlier today. The only thing I changed from the official installation document was using 

`sudo apt install gazebo`

Instead of

`sudo apt install gazebo11`

chemistwang, 2024-01-15:
Ubuntu 22.04 Jammy recommend Gazebo Harmonic [https://gazebosim.org/docs/all/getstarted](url) and there is an warning noted: gz-harmonic cannot be installed alongside gazebo-classic (eg. gazebo11). Try the new package.

D1Vincenzo, 2024-01-16:
I got the same problem, with same situation as yours. Still cannot find a solution

D1Vincenzo, 2024-01-18:
Are u using vm on mac with m1 chip? Thats what I did and I got same problem. After I setup same environment(ubuntu 22.04, ROS2 Humble) on another windows device(Intel chip), the problem never happens again.

chiadikav, 2024-01-30:
Im also on ARM mac and having the same issue wit Gazebo. no matter what, i can't get it to work. did you find a solution?

adolfo158, 2024-02-25:
Hi :-)   I have Ubuntu 23.04  Is there a Gazebo version that can be installed here ?

traversaro, 2024-02-26:
> Hi :-) I have Ubuntu 23.04 Is there a Gazebo version that can be installed here ?

Not from binaries, you can see the Gazebo Classic binaries officially available in Ubuntu repos at https://packages.ubuntu.com/search?suite=default&section=all&arch=any&keywords=gazebo&searchon=names , and as you can see `gazebo` (the package of gazebo classic) is only available up to Ubuntu 22.04 . 

For everyone else, can you please questions over in https://robotics.stackexchange.com/questions/tagged/gazebo ? Feel free to tag me there, but the issue tracker is more to track problems in the code, not for general questions. If you think instead there is a problem that needs to be tracked, feel free to comment here or create a new issue, thanks!

closed 2024-02-26 by traversaro

Saquib764, 2024-04-13:
> Are u using vm on mac with m1 chip or other arm chip? Thats what I did and I got same problem. And when I try it on my raspberry pi 4b, same thing happens.
> 
> However, after I setup same environment (ubuntu 22.04, ROS2 Humble) on another windows device(Intel chip), the problem never happens again.
> 
> The command I used to install is `curl -sSL http://get.gazebosim.org | sh` from https://classic.gazebosim.org/tutorials?tut=install_ubuntu.
> 
> If you did use arm, I also tried install another ubuntu 20.04 vm on my mac, and the gazebo can be installed normally as well. I guess problem is about gazebo11 on ubuntu 22.04 on ARM and thats why other people didnt have this problem.

I am also have exactly the same problem. I am using M1 pro and UTM to create at VM for Ubuntu 22.04

iampaulidrobo, 2024-06-01:
Did someone solved this?
I just recently upgraded my ubuntu to 24.04 from 20.04 and installed ros2 jazzy.And now I am held up for simulation testing of my code.

> <pre>sudo apt install gazebo
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package gazebo is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

<font color="#C01C28"><b>E: </b></font>Package &apos;gazebo&apos; has no installation candidate
</pre>

iampaulidrobo, 2024-06-01:
This solved for me:
https://gazebosim.org/docs/garden/ros_installation

homecircuit, 2024-11-12:
It is another alternative for solving the problem.
//--------------------------------------------------
sudo add-apt-repository ppa:openrobotics/gazebo11-gz-cli
sudo apt update
sudo apt-get install gazebo11
//--------------------------------------------------

Referent from:
https://gazebosim.org/docs/latest/install_gz11_side_by_side/