gazebosim/gazebo-classic #3358: protoc version mismatch building gazebo 11 from source in Ubuntu 22.04

john-maidbot, 2023-11-16:
## Environment
* OS Version: Ubuntu 22.04
* Source build

## Description
* Expected behavior: Following these instructions leads to a successful build of gazebo 11 http://classic.gazebosim.org/tutorials?tut=install_from_source&cat=install
* Actual behavior: build fails because the version of protobuf installed with libprotobuf-dev (4.23.3 a.k.a. 23.3) conflicts with the version used to build ignition-mgs5 (3.12), (both installed via apt)

## Steps to reproduce
1. Follow these instructions on Ubuntu 22.04 http://classic.gazebosim.org/tutorials?tut=install_from_source&cat=install 
2. gazebo version=11, ros_distro=humble

## Output
near the end of building gazebo_msgs, there is a lot of this error:
```
This file was generated by an older version of protoc which is incompatible with your Protocol Buffer headers. Please regenerate this file with a newer version of protoc.
```

I found that manually installing protobuf v3.12 from source fixed the problem and allowed me to build gazebo 11 successfully, but it seemed weird that the installed libprotobuf-dev version would conflict with ignition-msgs5.

roncapat, 2023-11-16:
I have the same exact problem. Available to test.

traversaro, 2023-11-16:
Ubuntu 22.04 ships with protobuf 3.12 apt package (see https://repology.org/project/protobuf/versions). How did you installed protobuf 4.23.3 a.k.a. 23.3 ? Can you report the output of the `apt list --installed` command,
and `CMakeCache.txt` contaned in the Gazebo build directory?

john-maidbot, 2023-11-17:
you are right @traversaro :open_mouth: 

I am now very confused about how i ended up with protobuf 4.23 installed in the first place :thinking: I double checked on a second computer and sure enough, only 3.12 of protobuf is installed from / available from apt.

I tried reverting my setup to how it was before I started the gazebo source install tutorial, but I was not able to reproduce the state I described. Maybe @roncapat can share their apt list and CMakeCache.txt, if they dont mind?

I do see that I appear to have two version of protoc though:
/usr/bin/protoc (which has version 3.12) and /usr/local/bin/protoc-23.3.0

I am starting to suspect it may have just been the result of a lingering dependency from another project i was working on. :thinking:

john-maidbot, 2023-11-17:
may the newer protoc in /usr/local/bin is somehow from pip. because pip appears to have many versions of protobuf newer than 3.12 available

traversaro, 2023-11-17:
> may the newer protoc in /usr/local/bin is somehow from pip. because pip appears to have many versions of protobuf newer than 3.12 available

What is the.output of pip list --verbose ?

john-maidbot, 2023-11-17:
atm its
```
pip list --verbose | grep proto
protobuf                             3.12.4                                                                                            /usr/lib/python3/dist-packages
```

but 
```
jjdang@jjdang:~/Desktop/Development/robot-meta/ros_ws/src/maidbot_desktop/maidbot_don_simulation/launch$ pip3 index versions protobuf
WARNING: pip index is currently an experimental command. It may be removed/changed in a future release without prior warning.
protobuf (4.25.1)
Available versions: 4.25.1, 4.25.0, 4.24.4, 4.24.3, 4.24.2, 4.24.1, 4.24.0, 4.23.4, 4.23.3, 4.23.2, 4.23.1, 4.23.0, 4.22.5, 4.22.4, 4.22.3, 4.22.1, 4.22.0, 4.21.12, 4.21.11, 4.21.10, 4.21.9, 4.21.8, 4.21.7, 4.21.6, 4.21.5, 4.21.4, 4.21.3, 4.21.2, 4.21.1, 3.20.3, 3.20.2, 3.20.1, 3.20.0, 3.19.6, 3.19.5, 3.19.4, 3.19.3, 3.19.2, 3.19.1, 3.19.0, 3.18.3, 3.18.1, 3.17.3, 3.17.2, 3.17.1, 3.17.0, 3.16.0, 3.15.8, 3.15.7, 3.15.6, 3.15.5, 3.15.4, 3.15.3, 3.15.2, 3.15.1, 3.15.0, 3.14.0, 3.13.0, 3.12.4, 3.12.2, 3.11.3, 3.11.2, 3.11.1, 3.11.0, 3.10.0, 3.9.2, 3.9.1, 3.9.0, 3.8.0, 3.7.1, 3.7.0, 3.6.1, 3.6.0, 3.5.2.post1, 3.5.2, 3.5.1, 3.5.0.post1, 3.4.0, 3.3.0, 3.2.0, 3.1.0.post1, 3.1.0, 3.0.0, 2.6.1, 2.6.0, 2.5.0, 2.4.1, 2.3.0, 2.0.3
  INSTALLED: 3.12.4
  LATEST:    4.25.1
```
the latest available version is 4.25.1. so i suspect i may have upgraded to a 4.XX version at some point. which was a bad idea :sweat_smile:

john-maidbot, 2023-11-17:
atm its
```
pip list --verbose | grep proto
protobuf                             3.12.4                                                                                            /usr/lib/python3/dist-packages
```

but 
```
 pip3 index versions protobuf
WARNING: pip index is currently an experimental command. It may be removed/changed in a future release without prior warning.
protobuf (4.25.1)
Available versions: 4.25.1, 4.25.0, 4.24.4, 4.24.3, 4.24.2, 4.24.1, 4.24.0, 4.23.4, 4.23.3, 4.23.2, 4.23.1, 4.23.0, 4.22.5, 4.22.4, 4.22.3, 4.22.1, 4.22.0, 4.21.12, 4.21.11, 4.21.10, 4.21.9, 4.21.8, 4.21.7, 4.21.6, 4.21.5, 4.21.4, 4.21.3, 4.21.2, 4.21.1, 3.20.3, 3.20.2, 3.20.1, 3.20.0, 3.19.6, 3.19.5, 3.19.4, 3.19.3, 3.19.2, 3.19.1, 3.19.0, 3.18.3, 3.18.1, 3.17.3, 3.17.2, 3.17.1, 3.17.0, 3.16.0, 3.15.8, 3.15.7, 3.15.6, 3.15.5, 3.15.4, 3.15.3, 3.15.2, 3.15.1, 3.15.0, 3.14.0, 3.13.0, 3.12.4, 3.12.2, 3.11.3, 3.11.2, 3.11.1, 3.11.0, 3.10.0, 3.9.2, 3.9.1, 3.9.0, 3.8.0, 3.7.1, 3.7.0, 3.6.1, 3.6.0, 3.5.2.post1, 3.5.2, 3.5.1, 3.5.0.post1, 3.4.0, 3.3.0, 3.2.0, 3.1.0.post1, 3.1.0, 3.0.0, 2.6.1, 2.6.0, 2.5.0, 2.4.1, 2.3.0, 2.0.3
  INSTALLED: 3.12.4
  LATEST:    4.25.1
```
the latest available version is 4.25.1. so i suspect i may have upgraded to a 4.XX version at some point. which was a bad idea :sweat_smile:

john-maidbot, 2023-11-20:
I would be ok with closing the issue since the problem seems to have been user error on my end. 

But while we are on the topic of versioning, it seems that gazebo 11 expects sdformat9 version >= 9.8, but ubuntu 22.04 only has sdformat9 version 9.7
https://repology.org/project/sdformat9/versions

maybe that will be fixed soon on the apt side.

traversaro, 2023-11-20:
> I would be ok with closing the issue since the problem seems to have been user error on my end.

I agree, as long as @roncapat can confirm this, I agree with closing.

> But while we are on the topic of versioning, it seems that gazebo 11 expects sdformat9 version >= 9.8, but ubuntu 22.04 only has sdformat9 version 9.7
https://repology.org/project/sdformat9/versions
> 
> maybe that will be fixed soon on the apt side.

Probably not. Packages in the upstream repositories of a released stable version of Ubuntu are only updated if there are security or bug fixes, and even in that case the procedure for the update is not trivial. If you want to compile the latest version of gazebo-classic on Ubuntu 22.04 with apt dependencies, I suggest that you also compile sdformat from source. This can be done relatively easily with vcstool + colcon and the following file: https://github.com/gazebo-tooling/gazebodistro/blob/master/gazebo11.yaml .

roncapat, 2023-11-20:
> > I would be ok with closing the issue since the problem seems to have been user error on my end.
> 
> I agree, as long as @roncapat can confirm this, I agree with closing.

For me, it still doesn't work on one machine of mine, but I tried another cleaner machine and managed to build. Maybe it's an "upgrade-from-ubuntu-20.04" problem or some mess done by me on this local configuration. So ok to close in my opinion.

closed 2023-11-20 by traversaro