gazebosim/gazebo-classic #3359: Very low Real Time Factor when using wideanglecamera

Juhyung-L, 2023-11-16:
OS: Ubuntu 22.04
Gazebo Version: 11

I have 3 cameras attached to a box (right, left, and back sides of the box).

When the camera type is `type="camera"`, the real time factor is around 0.6-0.7 
When I change the type to `type="wideanglecamera"`, the real time factor drops to ~0.1

I get that `wideanglecamera` captures a wider view and is more computationally expensive to simulate, but even when I set the FOV of the `wideanglecamera` to be the same as `camera`,(I set it to 
 90degrees/1.5708radians) the RTF is still ~0.1.

urdf for wideanglecamera
```
<?xml version="1.0"?>
<robot name="camera_test" xmlns:xacro="http://ros.org/wiki/xacro">
    <!-- define robot constants -->
    <xacro:property name="base_width" value="2"/>
    <xacro:property name="base_height" value="2"/>
    <xacro:property name="base_depth" value="2"/>

    <xacro:property name="camera_size" value="0.1"/>
    <xacro:property name="camera_zoff" value="0.3"/>

    <!-- define intertial property macros -->
    <xacro:macro name="box_inertia" params="m w h d">
        <inertial>
            <origin xyz="0 0 0" rpy="0 0 0"/>
            <mass value="${m}"/>
            <inertia ixx="${(m/12) * (h*h + d*d)}" ixy="0.0" ixz="0.0" iyy="${(m/12) * (w*w + d*d)}" iyz="0.0" izz="${(m/12) * (w*w + h*h)}"/>
        </inertial>
    </xacro:macro>

    <!-- wheelchair base -->
    <link name="base_link">
        <origin rpy="0 0 0 " xyz="0 0 0"/>
        <visual>
            <geometry>
                <box size="${base_width} ${base_height} ${base_depth}"/>
            </geometry>
            <material name="Cyan">
                <color rgba="0 1.0 1.0 1.0"/>
            </material>
        </visual>
        <collision>
            <geometry>
                <box size="${base_width} ${base_height} ${base_depth}"/>
            </geometry>
        </collision>
        <xacro:box_inertia m="10" w="${base_width}" h="${base_height}" d="${base_depth}"/>
    </link>

    <!-- cameras -->
    <xacro:macro name="camera_link" params="prefix x_reflect y_reflect angle_number">
        <link name="camera_${prefix}_link">
            <visual>
                <geometry>
                    <box size="${camera_size} ${camera_size} ${camera_size}"/>
                </geometry>
                <material name="Red">
                    <color rgba="1.0 0.0 0.0 1.0"/>
                </material>
            </visual>
            <collision>
                <geometry>
                    <box size="${camera_size} ${camera_size} ${camera_size}"/>
                </geometry>
            </collision>
            <xacro:box_inertia m="1" w="${camera_size}" h="${camera_size}" d="${camera_size}"/>
        </link>

[... 420 lines cut ...]
This is how I installed Gazebo in the contianer
```
##### Setup Gazebo #####
# essential install packages
RUN apt-get update && apt-get install -q -y --no-install-recommends \
    dirmngr \
    gnupg2 \
    lsb-release \
    binutils \
    mesa-utils \
    kmod \
    x-window-system \
    curl \
    && rm -rf /var/lib/apt/lists/*

# setup keys
RUN apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys D2486D2DD83DB69272AFE98867170598AF249743

# setup sources.list
RUN . /etc/os-release \
    && echo "deb http://packages.osrfoundation.org/gazebo/$ID-stable `lsb_release -sc` main" > /etc/apt/sources.list.d/gazebo-latest.list

# install gazebo packages
RUN curl -sSL http://get.gazebosim.org | sh
```

closed 2023-12-12 by Juhyung-L

Rudresh172, 2024-05-02:
Hey @Juhyung-L Did you find any solution to this? I am facing a similar issue. Although I can see the image in Rviz, it is very blurry, I have tried increasing the image resolution as well as the env_texture_size, but no changes.

Juhyung-L, 2024-05-22:
Hi, sorry for the late response.

My problem was not blurry images. It was that I was using 3 `wideanglecameras` in Gazebo, which was making the simulation very slow. I still have not found out why using `wideanglecameras` is slow. So I just decided to use the regular `camera`.