gazebosim/gazebo-classic #3360: The first time the GAZEBO is turned on, it opens quite hard

ionutnechita, 2023-11-18:
Hi Gazebo Team,

I have noticed many times, for the first time when I start Gazebo, on a newly created user, it opens quite hard. 
This causes the elements on the map not to be loaded for the first time.
After the first stop, for the second it starts very quickly, and the objects on the map are all loaded.

Maybe someone has seen this behavior before.

OS: Ubuntu 22.04 LTS
ROS: Humble
Gazebo SIM: 11.10.2+dfsg-1

traversaro, 2023-11-19:
What do you mean with "it opens quite hard"? The first time that is launched, Gazebo downloads the model database, but that should only add a delay on the first run, not break any existing simulation.

traversaro, 2023-11-19:
On a related note, can you follow the bug report template as described in https://github.com/gazebosim/gazebo-classic/issues/new?assignees=&labels=bug&projects=&template=bug_report.md ? This ensure that you provide all the necessary info to reproduce the problem.