gazebosim/gazebo-classic #3361: GAZEBO does not scale the map to the entire screen

ionutnechita, 2023-11-18:
Hi Gazebo Team,

I installed gazebo on my Ubuntu system today.
But i noticed that the map i want to use doesn't scale to full screen.

Maybe someone has seen this behavior before.

Photo:
[image]
[image]

gazebo -version
Gazebo multi-robot simulator, version 11.10.2
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

OS: Ubuntu 22.04 LTS
ROS: Humble
Gazebo SIM: 11.10.2+dfsg-1

traversaro, 2023-11-19:
Can you please follow the bug report template as described in https://github.com/gazebosim/gazebo-classic/issues/new?assignees=&labels=bug&projects=&template=bug_report.md ? This ensure that you provide all the necessary info to reproduce the problem, thanks!

renyuhui0415, 2024-03-27:
Have you solved it? I also have this problem, Ubuntu20.04 + gazebo11. After investigation, it was found that there was a problem with the display scaling of the Ubuntu system. When the scaling ratio was adjusted to 100%, it became normal.

ionutnechita, 2024-03-31:
Hi @renyuhui0415,

I didn't solve it.
I upgraded to Ubuntu 22.04.
But in the future I will use Ubuntu 24.04 or OpenSUSE.
I don't want to use X11 anymore.

renyuhui0415, 2024-04-01:
> Hi @renyuhui0415,
> 
> I didn't solve it. I upgraded to Ubuntu 22.04. But in the future I will use Ubuntu 24.04 or OpenSUSE. I don't want to use X11 anymore.

Hello, I have solved this problem.
I found a solution in other issues. Due to HidPI, I need to add the following two lines of code to **.bashrc** file.
```
export QT_AUTO_SCREEN_SCALE_FACTOR=0
export QT_SCREEN_SCALE_FACTORS=[1.0]
```
For a more detailed answer, please see this issue https://github.com/gazebosim/gazebo-classic/issues/2688

ManTang034, 2025-01-20:
You can try this one.
`export QT_SCALE_FACTOR=1.0`