gazebosim/gazebo-classic #3364: Mac gazebo11 upgrade and installation with brew throws error

paladincorners, 2024-02-01:
## Environment
* OS Version: MacOS
* Source or binary build? binary (brew)
  
  

## Description
* Expected behavior: 
Upgrading gazebo11 with `brew upgrade` should upgrade the package. Likewise clean install by following the directions should install without errors.

* Actual behavior: 
Received the following error during `brew upgrade` and also during `brew install gazebo11`:
Error: /usr/local/Cellar/gazebo11/11.14.0_8 is not a directory

## Steps to reproduce

1. Run `brew upgrade`
Or alternatively, if not installed yet, follow the installation instructions here:
https://classic.gazebosim.org/tutorials?tut=install_on_mac&cat=install#Alternativeinstallation:step-by-step

## Output

scpeters, 2024-02-21:
thanks for your report, but there isn't enough information for me to reproduce the problem. I just noticed a different problem with compiling gazebo11 on macOS (https://github.com/gazebosim/gazebo-classic/issues/3372). I'll patch the formula with that fix shortly, but after that I'd recommend trying your build again. If it still fails, can you report the failure at the osrf/homebrew-simulation repository and fill out the issue template? That will help us solve the issue together.

* https://github.com/osrf/homebrew-simulation/issues/new?assignees=&labels=bug&projects=&template=bug.md

TheKalbairCo, 2025-01-11:
Try this 
ulimit -n unlimited
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"
brew tap osrf/simulation
brew install gz-harmonic

TheKalbairCo, 2025-01-11:
try this 
ulimit -n unlimited
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"
brew tap osrf/simulation
brew install gz-harmonic