gazebosim/gazebo-classic #3366: Windows-vcpkg CI is failing February 2024

traversaro, 2024-02-01:
The windows-vcpkg CI is failing with error:
~~~
Building jasper[core,default-features,opengl]:x64-windows-release@4.0.0#2...
warning: -- Using community triplet x64-windows-release. This triplet configuration is not guaranteed to succeed.
-- [COMMUNITY] Loading triplet configuration from: D:\a\gazebo-classic\gazebo-classic\vcpkg\triplets\community\x64-windows-release.cmake
-- Installing port from location: D:\a\gazebo-classic\gazebo-classic\vcpkg\buildtrees\versioning_\versions\jasper\d7f51c26899ba9433a5a3ab92fc5b5887d5c764c
-- Downloading https://github.com/jasper-software/jasper/archive/402d096b39f4f618ad9e6ff2b4fc1b5eb260a2e4.tar.gz -> jasper-software-jasper-402d096b39f4f618ad9e6ff2b4fc1b5eb260a2e4.tar.gz...
-- Extracting source D:/a/gazebo-classic/gazebo-classic/vcpkg/downloads/jasper-software-jasper-402d096b39f4f618ad9e6ff2b4fc1b5eb260a2e4.tar.gz
-- Applying patch no_stdc_check.patch
-- Applying patch fix-library-name.patch
-- Using source at D:/a/gazebo-classic/gazebo-classic/vcpkg/buildtrees/jasper/src/5eb260a2e4-ca02e25be6.clean
-- Found external ninja('1.11.0').
-- Configuring x64-windows-release
-- Building x64-windows-release-rel
CMake Error at scripts/cmake/vcpkg_execute_build_process.cmake:134 (message):
    Command failed: "C:/Program Files/CMake/bin/cmake.exe" --build . --config Release --target install -- -v -j5
    Working Directory: D:/a/gazebo-classic/gazebo-classic/vcpkg/buildtrees/jasper/x64-windows-release-rel
    See logs for more information:
      D:\a\gazebo-classic\gazebo-classic\vcpkg\buildtrees\jasper\install-x64-windows-release-rel-out.log

Call Stack (most recent call first):
  installed/x64-windows-release/share/vcpkg-cmake/vcpkg_cmake_build.cmake:74 (vcpkg_execute_build_process)
  installed/x64-windows-release/share/vcpkg-cmake/vcpkg_cmake_install.cmake:16 (vcpkg_cmake_build)
  buildtrees/versioning_/versions/jasper/d7f51c26899ba9433a5a3ab92fc5b5887d5c764c/portfile.cmake:38 (vcpkg_cmake_install)
  scripts/ports.cmake:170 (include)

error: building jasper:x64-windows-release failed with: BUILD_FAILED
Elapsed time to handle jasper:x64-windows-release: 10 s
Please ensure you're using the latest port files with `git pull` and `vcpkg update`.
Then check for known issues at:
  https://github.com/microsoft/vcpkg/issues?q=is%3Aissue+is%3Aopen+in%3Atitle+jasper
You can submit a new issue at:
  https://github.com/microsoft/vcpkg/issues/new?title=[jasper]+Build+error+on+x64-windows-release&body=Copy+issue+body+from+D%3A%2Fa%2Fgazebo-classic%2Fgazebo-classic%2Fvcpkg%2Finstalled%2Fvcpkg%2Fissue_body.md
You can also submit an issue by running (GitHub CLI must be installed):
  gh issue create -R microsoft/vcpkg --title "[jasper] Build failure on x64-windows-release" --body-file D:/a/gazebo-classic/gazebo-classic/vcpkg/installed/vcpkg/issue_body.md
~~~

First failing build: https://github.com/gazebosim/gazebo-classic/actions/runs/7741924543
Last successful build (from ~3 months ago): https://github.com/gazebosim/gazebo-classic/actions/runs/6801639542

traversaro, 2024-02-01:
Probably this is a regression due to the change of GitHub Actions Windows image. @talregev are you interested in looking into this?

traversaro, 2024-02-01:
> We can try to update the vcpkg hash. maybe to the latest.

Are you interested in opening a PR for that?

talregev, 2024-02-01:
Already open:
https://github.com/gazebosim/gazebo-classic/pull/3367

talregev, 2024-02-01:
You can close this issue.

closed 2024-02-01 by traversaro