gazebosim/gazebo-classic #3372: Gazebo fails to compile against graphviz 10.0

scpeters, 2024-02-21:
## Environment
* OS Version: macOS 
* Source or binary build? source build of 11.14.0

[image] https://build.osrfoundation.org/job/gazebo-install-one_liner-homebrew-amd64/3068/

## Description
* Expected behavior: it should compile successfully with graphviz
* Actual behavior: it fails to compile

## Steps to reproduce

1. `brew install gazebo11`
2.
3.

## Output

The log in that CI job linked above does not include the relevant error, but I found the following when reproducing locally:

~~~
/tmp/gazebo11-20240220-52120-92epuq/gazebo-11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp:47:53: error: use of undeclared identifier 'TRUE'
    Agnode_t *node = agnode(_sgraph->graph(), NULL, TRUE);
                                                    ^
/tmp/gazebo11-20240220-52120-92epuq/gazebo-11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp:53:39: error: use of undeclared identifier 'TRUE'
    agsubnode(_sgraph->graph(), node, TRUE);
                                      ^
/tmp/gazebo11-20240220-52120-92epuq/gazebo-11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp:69:56: error: use of undeclared identifier 'TRUE'
            ("cluster_" + _name).toLocal8Bit().data(), TRUE);
                                                       ^
/tmp/gazebo11-20240220-52120-92epuq/gazebo-11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp:72:71: error: use of undeclared identifier 'TRUE'
        sgraph = agsubg(_sgraph->graph(), _name.toLocal8Bit().data(), TRUE);
                                                                      ^
make[2]: *** [gazebo/gui/CMakeFiles/gazebo_gui.dir/qgv/QGVSubGraph.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
~~~

Attempting to reproduce here:

* [image] https://build.osrfoundation.org/job/gazebo-ci-gazebo11-homebrew-amd64/198/

I am testing a patch to fix this.

scpeters, 2024-02-21:
fix in https://github.com/gazebosim/gazebo-classic/pull/3373

traversaro, 2024-02-21:
I confess that is nice for me when a new dependency first hit homebrew, so I get the fix for free when the same dependency gets updated on conda-forge. : )

closed 2024-02-21 by scpeters

scpeters, 2024-02-21:
> I confess that is nice for me when a new dependency first hit homebrew, so I get the fix for free when the same dependency gets updated on conda-forge. : )

happy to be of service :)

PR #3373 (2024-02-21): Fix build with graphviz 10.0
# 🦟 Bug fix

Fixes #3372.

## Summary

The `TRUE` macro has been removed in graphviz 10.0, which was recently released in homebrew-core. The fix is to replace `TRUE` with `1`.

To test: verify that the homebrew build compiles successfully.

## Checklist
- [X] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All [... cut]