gazebosim/gazebo-classic #3379: gazebo installation on Windows 11 (conda-forge) unsuccessful!

aiswaryame94, 2024-04-03:
Here are the steps I followed to install and run gazebo 11.14 on windows 11.
```
conda create -n gazebo_env
conda activate gazebo_env
conda install conda-forge::gazebo
gazebo --verbose
```
I get the following errors:
[image]
[image]

I also have paused the antivirus protection (Kaspersky), disabled windows firewall, toggled off the real-time protection on windows security. I also set gzserver.exe and gzclient.exe to use hig-processing NVIDIA graphics card,  in the NVIDIA control panel.

I am still facing the same error. Could you kindly help me solve the same?

traversaro, 2024-04-03:
Can you please report the output of conda info and conda list? Thanks!

ShadowDawg, 2024-04-06:
Hi, I've run into the same issue while installing gazebo 11.14.0. 
conda info: 
```
active environment : gazebo_env
    active env location : C:\Users\shado\miniconda3\envs\gazebo_env
            shell level : 2
       user config file : C:\Users\shado\.condarc
 populated config files : C:\Users\shado\.condarc
          conda version : 23.11.0
    conda-build version : not installed
         python version : 3.11.5.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=x86_64
                          __conda=23.11.0=0
                          __cuda=12.3=0
                          __win=0=0
       base environment : C:\Users\shado\miniconda3  (writable)
      conda av data dir : C:\Users\shado\miniconda3\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Users\shado\miniconda3\pkgs
                          C:\Users\shado\.conda\pkgs
                          C:\Users\shado\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\shado\miniconda3\envs
                          C:\Users\shado\.conda\envs
                          C:\Users\shado\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.11.0 requests/2.31.0 CPython/3.11.5 Windows/10 Windows/10.0.22000 solver/libmamba conda-libmamba-solver/23.12.0 libmambapy/1.5.3
          administrator : False
             netrc file : None
           offline mode : False
```

conda list:
```
# packages in environment at C:\Users\shado\miniconda3\envs\gazebo_env:
#
# Name                    Version                   Build  Channel
aom                       3.8.2                h63175ca_0    conda-forge
assimp                    5.3.1                h0dbab56_3    conda-forge
aws-c-auth                0.7.16               h7613915_8    conda-forge
aws-c-cal                 0.6.10               hf6fcf4e_2    conda-forge
aws-c-common              0.9.14               hcfcfb64_0    conda-forge
aws-c-compression         0.2.18               hf6fcf4e_2    conda-forge
aws-c-event-stream        0.4.2                h3df98b0_6    conda-forge
aws-c-http                0.8.1                h4e3df0f_7    conda-forge
aws-c-io                  0.14.6               hf0b8b6f_2    conda-forge
aws-c-mqtt                0.10.3               h96fac68_2    conda-forge
aws-c-s3                  0.5.4                h08df315_0    conda-forge
aws-c-sdkutils            0.1.15               hf6fcf4e_2    conda-forge
aws-checksums             0.1.18               hf6fcf4e_2    conda-forge
aws-crt-cpp               0.26.4               hbe739fa_2    conda-forge
aws-sdk-cpp               1.11.267             hfaf0dd0_4    conda-forge
azure-core-cpp            1.11.1               h249a519_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h91493d7_1    conda-forge
azure-storage-common-cpp  12.5.0               h91493d7_4    conda-forge
blosc                     1.21.5               hdccc3a2_0    conda-forge
bullet-cpp                3.25                 h2ab9e98_2    conda-forge
bzip2                     1.0.8                hcfcfb64_5    conda-forge
c-ares                    1.28.1               hcfcfb64_0    conda-forge
ca-certificates           2024.2.2             h56e8100_0    conda-forge
cairo                     1.18.0               h1fef639_0    conda-forge
cfitsio                   4.4.0                h9b0cee5_0    conda-forge
[... 244 lines cut ...]
ProgramW6432=C:\Program Files
PROJ_DATA=C:\Users\shado\miniconda3\envs\gazebo_env\Library\share\proj
PROJ_NETWORK=ON
PROMPT=(gazebo_env) $P$G
PSModulePath=C:\Users\shado\Documents\WindowsPowerShell\Modules;C:\Users\shado\AppData\Local\Google\Cloud SDK\google-cloud-sdk\platform\PowerShell
PUBLIC=C:\Users\Public
RlsSvcPort=22112
SDF_PATH=C:\Users\shado\miniconda3\envs\gazebo_env\Library\share\sdformat9\1.7
SESSIONNAME=Console
SSL_CERT_FILE=C:\Users\shado\miniconda3\Library\ssl\cacert.pem
SystemDrive=C:
SystemRoot=C:\Windows
TEMP=C:\Users\shado\AppData\Local\Temp
TMP=C:\Users\shado\AppData\Local\Temp
USERDOMAIN=SHADOW
USERDOMAIN_ROAMINGPROFILE=SHADOW
USERNAME=shado
USERPROFILE=C:\Users\shado
windir=C:\Windows
XML_CATALOG_FILES=file:///C:/Users/shado/miniconda3/envs/gazebo_env/etc/xml/catalog
ZES_ENABLE_SYSMAN=1
__CONDA_OPENSLL_CERT_FILE_SET="1"
```

But I've only faced this error intalling the latest version (11.14.0) and 11.13.0 runs fine!

traversaro, 2024-04-06:
Thanks, I was able to reproduce the problem.

traversaro, 2024-04-06:
The 0xc0000142 error means that some dll is not loaded, let's debug with https://github.com/lucasg/Dependencies .

traversaro, 2024-04-06:
Not even `gz` works.

traversaro, 2024-04-06:
Installing `gazebo==11.14.0=*_9` works fine, so it could be something related to boost or similar:

~~~
(gazebo) C:\Users\straversaro>mamba install gazebo==11.14.0=*_9

Looking for: ['gazebo==11.14.0[build=*_9]']

conda-forge/noarch                                  14.2MB @   7.2MB/s  2.0s
conda-forge/win-64                                  21.5MB @   7.3MB/s  3.0s

Pinned packages:
  - python 3.12.*
Transaction

  Prefix: C:\Users\straversaro\AppData\Local\miniforge3\envs\gazebo

  Updating specs:

   - gazebo==11.14.0[build=*_9]
   - ca-certificates
   - openssl

  Package             Version  Build        Channel           Size
--------------------------------------------------------------------
  Change:
--------------------------------------------------------------------

  - assimp              5.3.1  h0dbab56_3   conda-forge     Cached
  + assimp              5.3.1  h81f0834_2   conda-forge     Cached
  - gazebo            11.14.0  hb27d4c6_10  conda-forge     Cached
  + gazebo            11.14.0  h3902e0a_9   conda-forge       60MB

  Downgrade:
--------------------------------------------------------------------

  - libboost           1.84.0  hcc118f5_2   conda-forge     Cached
  + libboost           1.82.0  h65993cd_6   conda-forge     Cached
  - libboost-headers   1.84.0  h57928b3_2   conda-forge     Cached
  + libboost-headers   1.82.0  h57928b3_6   conda-forge     Cached
  - libboost-devel     1.84.0  h91493d7_2   conda-forge     Cached
  + libboost-devel     1.82.0  h91493d7_6   conda-forge     Cached

  Summary:

  Change: 2 packages
  Downgrade: 3 packages

  Total download: 60MB

--------------------------------------------------------------------

Confirm changes: [Y/n] y
gazebo                                              60.3MB @  12.3MB/s  4.9s

Downloading and Extracting Packages:

Preparing transaction: done
Verifying transaction: -
Executing transaction: done

traversaro, 2024-04-06:
> The 0xc0000142 error means that some dll is not loaded, let's debug with https://github.com/lucasg/Dependencies .

I do not see anything clearly wrong with that. I wonder if this problem is somehow related to https://github.com/robotology/gazebo-yarp-plugins/issues/678 .

traversaro, 2024-04-07:
The problem is present with all builds of libboost 1.84.0, so probably the problem is not related to some ABI break between different builds of libboost 1.84.0 . The problem is not present in Linux/macOS.

traversaro, 2024-04-08:
Some updates: https://github.com/conda-forge/gazebo-feedstock/pull/210#issuecomment-2043671771 : 

> I tested this locally, and it has the exact same problem of boost 1.84 builds, even if we are using boost 1.82 . On the other hand, I built locally gazebo-classic against 1.84, and it is working fine. At this point, I suspect that the difference between working and not working builds is the visual studio version, that got updated at some point between 1.82 and 1.84 builds.

First problematic build: https://github.com/conda-forge/gazebo-feedstock/runs/23137603666, compiled with MSVC 19.38.33135.0
Last successful build: https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=903757&view=logs&jobId=a70f640f-cc53-5cd3-6cdc-236a1aa90802 MSVC 19.38.33135.0

Ok, the MSVC is exactly the same. I wonder if it is something related to this comment https://github.com/conda-forge/conda-forge.github.io/issues/2102#issuecomment-2043560833 .

traversaro, 2024-04-08:
> Ok, ~the MSVC is exactly the same~ Edit, I was looking at the wrong job. Indeed the MSVC version is different, I wonder if it is something related to this comment [conda-forge/conda-forge.github.io#2102 (comment)](https://github.com/conda-forge/conda-forge.github.io/issues/2102#issuecomment-2043560833) .

Note that this (i.e. the use of vs2022) does not explain by itself the problem, it is probably some sort of regression in the vs2022 version used in the CI. However, for sure this explains why a downstream vs2019 build failed, see https://github.com/robotology/gazebo-yarp-plugins/pull/679 .

traversaro, 2024-04-08:
> The 0xc0000142 error means that some dll is not loaded, let's debug with https://github.com/lucasg/Dependencies .

For future reference: this was wrong. The dll not loaded/missing symbols error is 0xc0000135, 0xc0000142 is something like "application not starting correctly" or something similar.

h-vetinari, 2024-04-09:
If indeed this has something to do with vs2022, you should now be able to rerender & rebuild the respective feedstocks, and hopefully the problem will then disappear. Sorry for the trouble!

traversaro, 2024-04-09:
> If indeed this has something to do with vs2022, you should now be able to rerender & rebuild the respective feedstocks, and hopefully the problem will then disappear. Sorry for the trouble!

No problem! Indeed, I will open an issue to track the problem with vs2022, it is possible that just updating to a more recent version of vs2022 the problem disappears. Or let's hope that everyone stopped using gazebo classic and moved to gz-sim by the time that conda-forge migrates to vs2022.

traversaro, 2024-04-10:
The problem should be fixed with the `gazebo==11.14.0=orignameh160ba46_212` and `gazebo==11.14.0=gzcompatnameh1a9e25f_12` builds, i.e. now:

~~~
conda create -n gazebo gazebo
~~~

should work out of the box. Thanks a lot @aiswaryame94 and @ShadowDawg for reporting the problem, it was a tricky one. We also added tests in https://github.com/conda-forge/gazebo-feedstock/pull/211 to avoid these kind of regressions.

closed 2024-04-10 by traversaro

traversaro, 2024-08-13:
@USeebi can you open a new issue following the template? It is easier to track problems as new issue, as opposed to closed issues, thanks!