gazebosim/gazebo-classic #3382: Windows vcpkg job failing (April 2024

traversaro, 2024-04-13:
The windows-vcpkg job is failing:
~~~
2024-04-13T09:08:23.8245863Z Installing 100/216 freeimage:x64-windows-release@3.18.0#25...
2024-04-13T09:08:23.8246595Z Building freeimage:x64-windows-release@3.18.0#25...
2024-04-13T09:08:23.8249296Z D:\a\gazebo-classic\gazebo-classic\vcpkg\triplets\community\x64-windows-release.cmake: info: loaded community triplet from here. Community triplets are not built in the curated registry and are thus less likely to succeed.
2024-04-13T09:08:23.8253661Z D:\a\gazebo-classic\gazebo-classic\vcpkg\buildtrees\versioning_\versions\freeimage\423daf86bc904882a1fa9c68d56d56d7c52ac35f: info: installing overlay port from here
2024-04-13T09:08:23.8669701Z -- Downloading https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=cfhcable;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=pilotfiber;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=gigenet;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=versaweb;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=ayera;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=netactuate;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=phoenixnap;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=astuteinternet;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=freefr;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=netcologne;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=deac-riga;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=excellmedia;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=iweb;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=jaist;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=jztkft;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=managedway;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=nchc;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=netix;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=ufpr;https://sourceforge.net/projects/freeimage/files/Source%20Distribution/3.18.0/FreeImage3180.zip/download?use_mirror=tenet -> FreeImage3180.zip...
2024-04-13T09:08:25.1165043Z -- Extracting source D:/a/gazebo-classic/gazebo-classic/vcpkg/downloads/FreeImage3180.zip
2024-04-13T09:08:25.5702837Z -- Applying patch disable-plugins-depending-on-internal-third-party-libraries.patch
2024-04-13T09:08:25.5979682Z -- Applying patch use-external-jpeg.patch
2024-04-13T09:08:25.6232691Z -- Applying patch use-external-jxrlib.patch
2024-04-13T09:08:25.6479706Z -- Applying patch use-external-libtiff.patch
2024-04-13T09:08:25.6729853Z -- Applying patch use-external-openjpeg.patch
2024-04-13T09:08:25.6987108Z -- Applying patch use-external-png-zlib.patch
2024-04-13T09:08:25.7237928Z -- Applying patch use-external-rawlib.patch
2024-04-13T09:08:25.7497923Z -- Applying patch use-external-webp.patch
2024-04-13T09:08:25.7741505Z -- Applying patch use-external-openexr.patch
2024-04-13T09:08:25.7992990Z -- Applying patch use-freeimage-config-include.patch
2024-04-13T09:08:25.8251417Z -- Applying patch fix-function-overload.patch
2024-04-13T09:08:25.8544146Z -- Applying patch use-typedef-as-already-declared.patch
2024-04-13T09:08:25.8806100Z -- Applying patch use-functions-to-override-libtiff-warning-error-handlers.patch
2024-04-13T09:08:25.9037449Z -- Applying patch remove_auto_ptr.patch
2024-04-13T09:08:25.9373095Z -- Applying patch rawlib-build-fix.patch
2024-04-13T09:08:25.9635047Z -- Using source at D:/a/gazebo-classic/gazebo-classic/vcpkg/buildtrees/freeimage/src/eImage3180-3016f26dee.clean
2024-04-13T09:08:26.0683414Z -- Found external ninja('1.11.0').
2024-04-13T09:08:26.0710153Z -- Configuring x64-windows-release
2024-04-13T09:08:28.2539991Z CMake Error at scripts/cmake/vcpkg_execute_required_process.cmake:112 (message):
2024-04-13T09:08:28.2541656Z     Command failed: "C:/Program Files/Microsoft Visual Studio/2022/Enterprise/Common7/IDE/CommonExtensions/Microsoft/CMake/Ninja/ninja.exe" -v
2024-04-13T09:08:28.2543117Z     Working Directory: D:/a/gazebo-classic/gazebo-classic/vcpkg/buildtrees/freeimage/x64-windows-release-rel/vcpkg-parallel-configure
2024-04-13T09:08:28.2543895Z     Error code: 1
2024-04-13T09:08:28.2544138Z     See logs for more information:
2024-04-13T09:08:28.2545145Z       D:\a\gazebo-classic\gazebo-classic\vcpkg\buildtrees\freeimage\config-x64-windows-release-rel-CMakeCache.txt.log
2024-04-13T09:08:28.2546256Z       D:\a\gazebo-classic\gazebo-classic\vcpkg\buildtrees\freeimage\config-x64-windows-release-out.log
2024-04-13T09:08:28.2547408Z 
2024-04-13T09:08:28.2547546Z Call Stack (most recent call first):
2024-04-13T09:08:28.2548383Z   installed/x64-windows-release/share/vcpkg-cmake/vcpkg_cmake_configure.cmake:252 (vcpkg_execute_required_process)
2024-04-13T09:08:28.2549507Z   buildtrees/versioning_/versions/freeimage/423daf86bc904882a1fa9c68d56d56d7c52ac35f/portfile.cmake:45 (vcpkg_cmake_configure)
2024-04-13T09:08:28.2563658Z   scripts/ports.cmake:175 (include)
2024-04-13T09:08:28.2564073Z 
2024-04-13T09:08:28.2564080Z 
2024-04-13T09:08:28.2965251Z error: building freeimage:x64-windows-release failed with: BUILD_FAILED
2024-04-13T09:08:28.2966315Z Elapsed time to handle freeimage:x64-windows-release: 4.5 s
~~~

fyi @talregev

traversaro, 2024-04-13:
See https://github.com/gazebosim/gazebo-classic/actions/runs/8672412334/job/23782702381 .

talregev, 2024-04-13:
I will try to update the hash. Let's see if it do the trick.

PR #3383 (2024-04-14): Update vcpkg hash
Fix #3382
Change to windows-2022
~~Update vcpkg hash~~
Update latest cmake
closed 2024-04-14 by traversaro