gazebosim/gazebo-classic #3384: gazebo11 error on macos 14

mengchaoheng, 2024-04-15:
## Environment
* OS Version: macos 14.3.1
* Source: 
```Console
➜  ~ gazebo --version
Gazebo multi-robot simulator, version 11.14.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

```

## Description
* Expected behavior: px4_sitl run 
* Actual behavior: 
```Console
 PX4-Autopilot git:(heads/v1.13.3) ✗ make px4_sitl gazebo           
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/tmp && /Users/mch/Proj/Ma...sers/mch/Proj/Mac_DF/PX4-Autopilot /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_defaul
SITL ARGS
sitl_bin: /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /Users/mch/Proj/Mac_DF/PX4-Autopilot
build_path: /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH :/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
Using: /Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_gazebo/models/iris/iris.sdf
Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
gzserver(24755,0x204aa5240) malloc: Incorrect checksum for freed object 0x7fc70c1af470: probably modified after being freed.
Corrupt value: 0x4d555458
gzserver(24755,0x204aa5240) malloc: *** set a breakpoint in malloc_error_break to debug
/Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_run.sh: line 217: 24755 Abort trap: 6           gzserver $verbose $world_path $ros_args
SITL COMMAND: "/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/Users/mch/Proj/Mac_DF/PX4-Autopilot"/test_data
Creating symlink /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/etc -> /Users/mch/Proj/Mac_DF/PX4-Autopilo

_____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
* CAL_MAG0_ID: curr: 0 -> new: 197388
* CAL_MAG1_ID: curr: 0 -> new: 197644
* SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* SYS_AUTOCONFIG: curr: 1 -> new: 0
* IMU_INTEG_RATE: curr: 200 -> new: 250
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
Gazebo multi-robot simulator, version 11.14.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
[Err] [ConnectionManager.cc:121] Failed to connect to master in 30 seconds.
[Err] [gazebo_shared.cc:78] Unable to initialize transport.
[Err] [gazebo_client.cc:56] Unable to setup Gazebo

```

## Steps to reproduce

 [Environment Setup](https://docs.px4.io/main/en/dev_setup/dev_env_mac.html) step by step

## Output
```Console
Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
gzserver(24755,0x204aa5240) malloc: Incorrect checksum for freed object 0x7fc70c1af470: probably modified after being freed.
Corrupt value: 0x4d555458
gzserver(24755,0x204aa5240) malloc: *** set a breakpoint in malloc_error_break to debug
/Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_run.sh: line 217: 24755 Abort trap: 6           gzserver $verbose $world_path $ros_args
```

see https://github.com/PX4/PX4-Autopilot/issues/22826

mengchaoheng, 2024-04-20:
I have fix https://github.com/PX4/PX4-Autopilot/issues/22826#issuecomment-2067587623

palahmd, 2024-05-15:
Your fix didn't work for me. Does anyone else have the same issue? I've your instructions and also installing an older version of gazebo11.