gazebosim/gazebo-classic #3390: General guide on building gazebo from source on Windows 11

WillyTuring, 2024-08-02:
Hi you guys,

I am developing our custom UI for interacting with 6 dof robotic arm based on Gazebo in wsl2 on Windows11; the primary part is almost finished, now we want to migrate the application to the native Windows OS, since we have done a few modifications on the gazebo source code, we want to build the application from source as well on Windows rather than using Conda or direct gazebo installation with "vcpkg install --triplet=x64-windows gazebo".

The problem we have encountered is that since the CMakeLists.txt file from "https://github.com/gazebosim/gazebo-classic/tree/gazebo11" is intended for the build up on Unix-like system with such as pkg-config tool etc; we have changed a lot to suit the CMake on Windows, but still run into a lot of dependencies problem like Dart package, profiler, tcmalloc just to name a few.

Now my question is that has anyone who has built it from source on Windows, will it work? Is there any general guidance on how to do that? Any suggestions would be really appreciated, thanks in advance.

traversaro, 2024-08-02:
vcpkg and conda packages are built from source on Windows, so I can confirm that it is possible to build Gazebo from source on Windows. 

Furthermore, also the CI builds from source on Windows, installing depedencies using conda-forge (https://github.com/gazebosim/gazebo-classic/blob/e4b4d0fb752c7e43e34ab97d0e01a2a3eaca1ed4/.github/workflows/conda-forge.yml#L97-L102) or vcpkg (https://github.com/gazebosim/gazebo-classic/blob/e4b4d0fb752c7e43e34ab97d0e01a2a3eaca1ed4/.github/workflows/windows-config/action.yml#L40-L54). So I would say that in general indeed someone build from source Gazebo. 

Indeed, Gazebo Classic has a lot of dependencies, so probably it would be easier to help you if you specified how you are installing these dependencies. Just to start from the first point you make:

> is intended for the build up on Unix-like system with such as pkg-config tool etc

In general, `pkg-config` works fine on Windows. What problem do you have with pkg-config? Back in time indeed I made a lot of fixes to ensure that Gazebo Classic and its dependencies configured and compiled fine even without pkg-config mainly as vcpkg CI back in time did not installed pkg-config, but since vcpkg CI now installed pkg-config since a few years, I doubt that is still working fine.

traversaro, 2024-08-02:
> The actual problem with pkg-config tool in my case is since we are installing all those dependency packages using vcpkg rather than using pkg-config for installation, thus there is no .pc file for it to locate; rather we use find_package() to locate those packages.

I am afraid I am not following, sorry! vcpkg has really good pkg-config support, it is even one of the few package managers (that I know of) that actually validates the pkg-config dependency graph, see https://github.com/microsoft/vcpkg/blob/b8ac6696e3af59f4f0666ab81a48ae589cde00a8/scripts/cmake/vcpkg_fixup_pkgconfig.cmake#L115-L136 . 

As I mentioned, in the CI we are building gazebo against vcpkg-provided dependencies (see https://github.com/gazebosim/gazebo-classic/tree/gazebo11/.github/workflows/windows-config and https://github.com/gazebosim/gazebo-classic/tree/gazebo11/.github/workflows/windows-config). @talregev updated the latest hash in https://github.com/gazebosim/gazebo-classic/pull/3367, so it is perfectly possible that there was regressions in recent versions of vcpkg, but for that is easier if you precisely report how are you trying to compile gazebo (for example, which version of vcpkg you are trying to use, etc etc) and the exact errors you are obtaining.

traversaro, 2024-08-02:
By the way, based on https://github.com/microsoft/vcpkg/issues/17481 it seems that installing pkgconf port in vcpkg should be enough to get pkg-config on Windows.

traversaro, 2024-08-02:
> The problem with vcpkg i have currently is that there are a lot of packages just don't exist

Again, can you be a bit more detailed? Gazebo is able to compile fine with the depedencies listed in https://github.com/gazebosim/gazebo-classic/blob/gazebo11_11.14.0/deps/vcpkg/manifest/vcpkg.json#L7-L56, what dependencies are you missing?

talregev, 2024-08-02:
You can compile dependencies of classic gazebo with vcpkg on windows.
You need a specific hash. Newer hash will not work. Also you need to do it with manifest mode (with vcpkg.json) because it depend on older versions of libraries. The only way to do it, it with manifest mode.
You can check the Yaml of the ci to see how it done.

talregev, 2024-08-02:
> Again, can you be a bit more detailed? Gazebo is able to compile fine with the depedencies listed in [...]

Also override to the older version:
https://github.com/gazebosim/gazebo-classic/blob/gazebo11_11.14.0/deps/vcpkg/manifest/vcpkg.json#L140-L149

talregev, 2024-08-02:
Hash found here:
https://github.com/gazebosim/gazebo-classic/blob/gazebo11_11.14.0/deps/vcpkg/manifest/vcpkg.json#L6

WillyTuring, 2024-08-02:
> Again, can you be a bit more detailed? Gazebo is able to compile fine with the depedencies listed in [...]

[image]
This screenshot doesn't tell the whole story since i am modifying the SearchForStuff.cmake file under gazebo/cmake/.. directory to ditch using pkg-config tool as a whole, and as i have stated earlier since all the dependencies are installed with vcpkg rather than pkg-config, there is only config.cmake file no .pc file hence we are doing it manually. 

But as i have also stated, the problem with that is we need to manually download each of the dependency package wtih vcpkg and modify the .cmake file to locate them one by one, it is tedious and not systematic in my view.

But you mentioned the port for pkgconfig, it will be a much more elegant approach if it works.

talregev, 2024-08-02:
pc files in vcpkg are located in "vcpkg\installed\x64-windows\lib\pkgconfig"

WillyTuring, 2024-08-02:
> pc files in vcpkg are located in "vcpkg\installed\x64-windows\lib\pkgconfig"

Sorry for my inexperience, i found out just now, have been using CMake tool to find the config.cmake file from \vcpkg\installed\x64-windows\share directory the whole time.

talregev, 2024-08-02:
if you install it with cmake with vcpkg tool chain (-DCMAKE_TOOLCHAIN_FILE=vcpkg\scripts\buildsystems\vcpkg.cmake) it will detect correctly the pkg config exe and the find the pc if available.

WillyTuring, 2024-08-02:
> if you install it with cmake with vcpkg tool chain (-DCMAKE_TOOLCHAIN_FILE=vcpkg\scripts\buildsystems\vcpkg.cmake) it will detect correctly the pkg config exe and the find the pc if available.

Yes, i understand now. Since you mentioned about the approach using specific hash and build it with manifest mode to solve the dependency issue, i think this is a more elegant and proven approach, i will try that now to see how it goes. Will come back to report the results.

WillyTuring, 2024-08-02:
"E:\19.gazebo_project\gazebo>git status
On branch gazebo11
Your branch is up to date with 'origin/gazebo11'.

nothing to commit, working tree clean

E:\19.gazebo_project\gazebo>git checkout db0473513e5dc73ec6b6f431ff05d2f398eea042
fatal: unable to read tree (db0473513e5dc73ec6b6f431ff05d2f398eea042)"

Is this the correct hash? Or did i miss somthing?

talregev, 2024-08-02:
This hash is for vcpkg. not gazebo.
Gazebo should be good on the latest default branch.

WillyTuring, 2024-08-02:
> This hash is for vcpkg. not gazebo. Gazebo should be good on the latest default branch.

Dame me.

talregev, 2024-08-02:
try to use manifest mode. it will do it automatically for you

WillyTuring, 2024-08-02:
[image]
I am dealing with this error for a while now, i have necessary VC components installed, any idea why?

WillyTuring, 2024-08-02:
[image]
This command just runs perfectly.

WillyTuring, 2024-08-02:
[image]
We have tested out with the newest vcpkg, the install command works just fine.

Whereas for the specific hash version, not so much.
[image]

WillyTuring, 2024-08-02:
[image]
I think all the necessary components are there. What did i miss...

traversaro, 2024-08-02:
I think I had a similar issue in one project, I worked it around using a newer vcpkg tool, I am currently away from a laptop, as soon as I am back I can provide more details.

WillyTuring, 2024-08-03:
> I think I had a similar issue in one project, I worked it around using a newer vcpkg tool, I am currently away from a laptop, as soon as I am back I can provide more details.

I have tried the whole morning, if using the newer version of vcpkg tool, we sort of go back to the original state where there are quite a lot of dependencies missing and need to install manually.

talregev, 2024-08-03:
@WillyTuring From your screenshot, you don't use manifest mode for vcpkg.
On vcpkg you should be on the latest master. Run `bootstrap-vcpkg.bat` to be on the latest tool of vcpkg.
Then when you do the install, first do the manifest mode install:
These commands is done on gazebo repo. (You can adopt them later for your needs). 
Note that manifest mode will remove any previous installation, that why you will need to do it, in 2 steps.

Note that manifest not change the hash of the git, it take the ports from old hash as it specify in the manifest file.

Note I change from windows-release to regular windows triplet. it will compile both debug and release. It will take more time.
I change this for you, because I don't know if the other depends you want to install, they will install correctly on windows-release triplet. 

```
cd <paht to gazebo repo>
git clone vcpkg
cd vcpkg
bootstrap-vcpkg.bat
vcpkg install  --x-manifest-root=..\deps\vcpkg\manifest --x-install-root=installed --host-triplet=x64-windows --clean-after-build
vcpkg install <your missing dependencies>
``` 
Then you need to compile gazebo without manifest mode, because you already have the dependencies you want,  and manifest mode will delete the other dependencies you install.

```
"C:\ProgramData\Microsoft\Windows\Start Menu\Programs\Visual Studio 2022\Visual Studio Tools\VC\x64 Native Tools Command Prompt for VS 2022.lnk"
cd <path to your gazebo repo>
        cmake . -G Ninja -B build ^
            -DCMAKE_BUILD_TYPE=Release ^
            -DCMAKE_TOOLCHAIN_FILE=vcpkg\scripts\buildsystems\vcpkg.cmake ^
            -DVCPKG_OVERLAY_TRIPLETS=deps\vcpkg\triplets ^
            -DVCPKG_INSTALLED_DIR=%CD%\vcpkg\installed ^
            -DVCPKG_TARGET_TRIPLET=x64-windows^
            -DVCPKG_HOST_TRIPLET=x64-windows^
            -DVCPKG_INSTALL_OPTIONS=--clean-after-build ^
            -DUSE_EXTERNAL_TINYXML=ON ^
            -DUSE_EXTERNAL_TINYXML2=ON ^
            -DUSE_EXTERNAL_TINY_PROCESS_LIBRARY=ON ^
            -DGZ_PROTOBUF_USE_CMAKE_CONFIG=ON ^
            -DCMAKE_INSTALL_PREFIX=%CD%\vcpkg\installed\x64-windows
```

I will not available in the weekend.
I didn't check this command for a while, and also changed on the fly for your case.
And not check these commands after the changes. might be a small errors. 
This should be a guide. and you can take it from here and try and workout the things you need.

Hope that will help you for what you need.

WillyTuring, 2024-08-03:
[image]
Here we used the specific hash version of vcpkg and ran .bat file.

[image]
We have the vcpkg.json file located at \gazebo_project\gazebo.

[image]
We then install those dependencies from .json file on manifest mode at the directory where .json file located at.

Below is the output if we use the latest vcpkg tool rather than that specified hash version, we got some dependency errors.
[image]

talregev, 2024-08-03:
please be on latest vcpkg master and try to follow my guide.

WillyTuring, 2024-08-03:
[image]
We are on the latest vcpkg, then we encountered the following error:

[image]

talregev, 2024-08-03:
Can you record all the steps from my guide and show the status include the commands itself?
There is a reason why don't work on the latest dependencies because it break classic gazebo.

WillyTuring, 2024-08-03:
step 1:
[image]

step 2:
[image]

step 3:
[image]

step 4:
[image]
...
[image]

talregev, 2024-08-03:
This is the wrong manifest file. you should take it from gazebo repo

WillyTuring, 2024-08-03:
[image]
I did.

Only when i encountered the dependency problem i added with:
[image]

Even if i added those two dependency lines, the same error persists.

talregev, 2024-08-03:
also don't put "." on vcpkg root manifest. please specify a dir that it take the correct manifest. because it can take from vcpkg itself.

WillyTuring, 2024-08-03:
I has taken a good 3h approximately without failing at the moment; i don't think it will finish soon at moment...thought i am going to bed and check it out tomorrow morning; how long does this thing usually take though in general?

WillyTuring, 2024-08-04:
Update: the dependency lists installed in manifest mode is succeed.

The new problem happens in the cmake . step:
[image]

But we actuall have the necessary libs here:
[image]

Seems like the cmake just couldn't find it? Why?

WillyTuring, 2024-08-04:
Start to wonder there are still some problems with dependency installation, so went back to re-run the command again:
[image]

Got the error this time:
[image]

What on earth.

WillyTuring, 2024-08-04:
The latest build update situation:
[image]

We have Ogre libraries installed "E:\19.gazebo_project\gazebo\installed" which is the specified path when we install the dependencies using manifest mode.
But what we have found out is that not all libraries will have the ".pc" file generated along with it, ogre is this case.
So in SearchForStuff.cmake file, since it uses a lot of pkg-config commands,  we got error:
[image]
...
[image]

traversaro, 2024-08-05:
> Start to wonder there are still some problems with dependency installation, so went back to re-run the command again: [...]

The relevant error is in the `install-x64-windows-dbg-out.log` .

WillyTuring, 2024-08-05:
I am wondering if the problem that some of the dependencies don't have .pc file installed is because their portfile doesn't get invoked? Here is our test speculation:
[image]

Here are the procedures:
1.We first remove the ogre dependency by editing the manifest file;
2.Verify its removal through "E:\19.gazebo_project\gazebo\installed\x64-windows\share" and "E:\19.gazebo_project\gazebo\installed\x64-windows\lib".
3.Then added ogre dependency in to the manifest file, run the intall command again "E:\19.gazebo_project\vcpkg\vcpkg.exe install --x-manifest-root=E:\19.gazebo_project\gazebo --x-install-root=E:\19.gazebo_project\gazebo\installed --host-triplet=x64-windows --clean-after-build"
4.The result: we didn't see anything we added in the portfile gets printed out in the terminal.

WillyTuring, 2024-08-05:
> The relevant error is in the `install-x64-windows-dbg-out.log` .

The dependency installation problem is solved now.
The problem now is the build process with command "cmake . -G Ninja -B build ^ -DCMAKE_BUILD_TYPE=Release ^ -DCMAKE_TOOLCHAIN_FILE=..\vcpkg\scripts\buildsystems\vcpkg.cmake ^ -DVCPKG_OVERLAY_TRIPLETS=..\vcpkg\triplets ^ -DVCPKG_INSTALLED_DIR=..\gazebo\installed ^ -DVCPKG_TARGET_TRIPLET=x64-windows ^ -DVCPKG_HOST_TRIPLET=x64-windows ^ -DVCPKG_INSTALL_OPTIONS=--clean-after-build ^ -DUSE_EXTERNAL_TINYXML=ON ^ -DUSE_EXTERNAL_TINYXML2=ON ^ -DUSE_EXTERNAL_TINY_PROCESS_LIBRARY=ON ^ -DGZ_PROTOBUF_USE_CMAKE_CONFIG=ON ^ -DCMAKE_INSTALL_PREFIX=..\gazebo\installed\x64-windows ^ -DCMAKE_MAKE_PROGRAM="C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\Common7\IDE\CommonExtensions\Microsoft\CMake\Ninja\ninja.exe" ^ -DCMAKE_C_COMPILER="C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.40.33807\bin\Hostx64\x64\cl.exe" ^ -DCMAKE_CXX_COMPILER="C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.40.33807\bin\Hostx64\x64\cl.exe""

WillyTuring, 2024-08-05:
Sorry here is the command to make it more readable:
[image]

traversaro, 2024-08-05:
Can you report the `CMakeCache.txt` of Gazebo? It seems that is not finding `pkg-config`.

WillyTuring, 2024-08-05:
Sure, we have it installed, unless CMake didn't use it though:
[image]

WillyTuring, 2024-08-05:
[CMakeCache.txt](https://github.com/user-attachments/files/16492216/CMakeCache.txt)

traversaro, 2024-08-05:
Apparently CMake is using vcpkg's pkgconf.exe, see the lines:

~~~
//Arguments to supply to pkg-config
PKG_CONFIG_ARGN:STRING=

//pkg-config executable
PKG_CONFIG_EXECUTABLE:FILEPATH=E:/19.gazebo_project/gazebo/installed/x64-windows/tools/pkgconf/pkgconf.exe
~~~

So I would try to make sure if that can find OGRE, that is the output of:

~~~
E:/19.gazebo_project/gazebo/installed/x64-windows/tools/pkgconf/pkgconf.exe --list-all
~~~

Even if probably we need to also set PKG_CONFIG_PATH env variable to point to vcpkg directories.

WillyTuring, 2024-08-05:
If you also look at this:
[image]

They are the result from SearchForStuff.cmake which is invoked from CMakeLists.txt file.

And in here we could see it uses a lot of pkg-config tool related commands like pkg_check_modules() to locate the libraries which searching for ".pc" file automatically; but for some reasons, when we use the vcpkg install command under manifest mode, a lot of ".pc" files just didn't get installed alongside with standard cmake library(.lib), this is what i meant the portfiles didn't seem to be invoked when installing the dependencies.

traversaro, 2024-08-05:
> And in here we could see it uses a lot of pkg-config tool related commands like pkg_check_modules() to locate the libraries which searching for ".pc" file automatically; but for some reasons, when [...]

That is strange, can you report which packages in not installing .pc files, i.e. where those files are expected to be installed and what files are installed? One unusual thing of your setup that I guess it not tested in CI is that you have a `.` in your install path, not sure if that can confused pkg-config somehow.

WillyTuring, 2024-08-05:
Nah, the installation seems to be correct i reckon, here is the command:
"E:\19.gazebo_project\vcpkg\vcpkg.exe install --x-manifest-root=E:\19.gazebo_project\gazebo --x-install-root=E:\19.gazebo_project\gazebo\installed --host-triplet=x64-windows --clean-after-build"

I have not used "." in the path since [talregev](https://github.com/talregev) pointed out.

I did use relative path in my build "cmake ." command though as following:
[image]

traversaro, 2024-08-05:
Ok, if the install is correct and the `.pc` files are correctly, can you indeed check if `pkgconf.exe` can find the packages?

WillyTuring, 2024-08-05:
"That is strange, can you report which packages in not installing .pc files, i.e. where those files are expected to be installed and what files are installed?"
——I want to see a lof of them as you can see from the screenshot above.

But i only tested ogre package.
Under "E:\19.gazebo_project\gazebo\installed\x64-windows\lib", we could find ogre*.lib; but under "E:\19.gazebo_project\gazebo\installed\x64-windows\lib\pkgconfig" there is no corresponding .pc files.

WillyTuring, 2024-08-05:
No, what do i mean is that the installation for standard cmake library is correct(aka .lib files are being installed correctly); but not the corresponding .pc file, they weren't being installed correctly during the installation process.

traversaro, 2024-08-05:
> Under "E:\19.gazebo_project\gazebo\installed\x64-windows\lib", we could find ogre*.lib; but under "E:\19.gazebo_project\gazebo\installed\x64-windows\lib\pkgconfig" there is no corresponding .pc files.

Is the folder `E:\19.gazebo_project\gazebo\installed\x64-windows\lib\pkgconfig` totally empty or just some .pc files are missing? Which .pc files are missing?

WillyTuring, 2024-08-05:
[pkgconfig_list.txt](https://github.com/user-attachments/files/16492703/pkgconfig_list.txt)

I am going to be away from the laptop for a while, gonna come back soon.

WillyTuring, 2024-08-05:
@traversaro Any update thoughts?

WillyTuring, 2024-08-05:
Man... I still can not figure out the solution, when did you guys past the CI/CD test?

WillyTuring, 2024-08-05:
Sorry sometimes the text is just too long, if so next time i will upload in the format of .txt file.

command: .\vcpkg.exe list --x-install-root=E:\19.gazebo_project\gazebo\installed
[image]
[package_list.txt](https://github.com/user-attachments/files/16503515/package_list.txt)

talregev, 2024-08-06:
you are using latest ogre. it will not compile with classic gazebo. 
In the manifest we lock it to 1.12
https://github.com/gazebosim/gazebo-classic/blob/e4b4d0fb752c7e43e34ab97d0e01a2a3eaca1ed4/deps/vcpkg/manifest/vcpkg.json#L143

WillyTuring, 2024-08-06:
1.Here is the vcpkg.json file; i think i haven't changed anything.
[vcpkg.json](https://github.com/user-attachments/files/16505305/vcpkg.json)

2.Here is the command.
"E:\19.gazebo_project\gazebo>E:\19.gazebo_project\vcpkg\vcpkg.exe install --x-manifest-root=E:\19.gazebo_project\gazebo --x-install-root=E:\19.gazebo_project\gazebo\installed --host-triplet=x64-windows --clean-after-build"

3.Here is the installation output
[install_out.txt](https://github.com/user-attachments/files/16505324/install_out.txt)
[image]

talregev, 2024-08-06:
try this command first, and show the results:
```
vcpkg install  --x-manifest-root=..\deps\vcpkg\manifest --x-install-root=installed --host-triplet=x64-windows --clean-after-build
```

If first commands not failed show this:
```
vcpkg list ogre
```

It should show you ogre 1.12.9

talregev, 2024-08-06:
> "--x-manifest-root=..\deps\vcpkg\manifest" ——But there is no manifest file in this directory.

adapt the command on your pc.

WillyTuring, 2024-08-06:
> E:\19.gazebo_project\vcpkg\vcpkg.exe install --x-manifest-root=E:\19.gazebo_project\gazebo --x-install-root=E:\19.gazebo_project\gazebo\installed --host-triplet=x64-windows --clean-after-build

I think this command it is 
"E:\19.gazebo_project\vcpkg\vcpkg.exe install --x-manifest-root=E:\19.gazebo_project\gazebo --x-install-root=E:\19.gazebo_project\gazebo\installed --host-triplet=x64-windows --clean-after-build"

talregev, 2024-08-06:
> I think this command it is "E:\19.gazebo_project\vcpkg\vcpkg.exe install --x-manifest-root=E:\19.gazebo_project\gazebo --x-install-root=E:\19.gazebo_project\gazebo\installed [...]

ok. try this.

Note:
My commands is over gazebo classic repo after you clone it, and also clone vcpkg repo and you are inside it.
On gazebo classic repo, the manifest is on that folder. 
https://github.com/gazebosim/gazebo-classic/tree/gazebo11/deps/vcpkg/manifest

[63 more replies]

closed 2024-08-12 by WillyTuring