gazebosim/gazebo-classic #3392: Loading .dll plugin on Windows problem

WillyTuring, 2024-08-12:
Does anyone have any experience on loading the .dll plugin on Windows?

The problem i am encountering now is that the "this->node->Init();" the node could be initiated but when we call Advertise() method trying to public a topic, the program hangs there.

traversaro, 2024-08-12:
> Does anyone have any experience on loading the .dll plugin on Windows?

Which `.dll plugin` ?

WillyTuring, 2024-08-12:
[image]
Here in my world.world file, i am loading a GUI plugin.

Here is the initial part of the code:
[image]

The Node init succeed, but we hand in calling Advertise() method, here is the verbose printout:
[image]

WillyTuring, 2024-08-13:
[image]
Hi any idea about this problem?

> > Does anyone have any experience on loading the .dll plugin on Windows?
> 
> Which `.dll plugin` ?

I have read the comment on this issue ""File already exists in database: vector3d.proto" when World::Load is called #3225 ; but didn't quite understand how to apply it.

traversaro, 2024-08-13:
Are you sure you are not compiling some library as static somewhere? That error is related to a proto being added to times to the internal protobuf database.

WillyTuring, 2024-08-13:
I am using the custom message proto for exchange data among the plugins(GUI plugin, model plugin as well as the visual plugin); i compile them as STATIC library and link it as such:
[image]

What is the issue with that?

WillyTuring, 2024-08-13:
> I am using the custom message proto for exchange data among the plugins(GUI plugin, model plugin as well as the visual plugin); i compile them as STATIC library and link it as such: [...]

I could build it as SHARED library, but it didn't change the result.

WillyTuring, 2024-08-13:
Why is it that when trying to build a SHARED library like below:
[image]
only .dll file is being generated, but no .lib file?

traversaro, 2024-08-13:
> Why is it that when trying to build a SHARED library like below:
> [image]
> only .dll file is being generated, but no .lib file?

Probably you are not exporting any symbol, either with dllexport decorators or via CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS.

WillyTuring, 2024-08-13:
when i use "set(CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS TRUE)" to force it out, the library it generates being linked by other plugins will crash the gazebo, meaning that once launching gazebo, after a while, it will exit.

Will dllexport decorator give us different result? Btw since we are grouping several custom .proto message to one library trying to solve the error of "File already exists in database", i am not sure if the dllexport can still work.

traversaro, 2024-08-13:
> Will dllexport decorator give us different result?

I guess it depends on the nature of the crash, it is difficult to answer in general.

WillyTuring, 2024-08-13:
> > Will dllexport decorator give us different result?
> 
> I guess it depends on the nature of the crash, it is difficult to answer in general.

No error being reported whatsoever, the gazebo application just exit out.

traversaro, 2024-08-13:
> > > Will dllexport decorator give us different result?
> > 
> > 
> > I guess it depends on the nature of the crash, it is difficult to answer in general.
> 
> No error being reported whatsoever, the gazebo application just exit out.

Even if that case, the answer depends on why it crashed. Even if you do not have any error, you can investigate the nature of the crash by attaching a debugger to the process and obtaining a backtrace.

WillyTuring, 2024-08-13:
> > Will dllexport decorator give us different result?
> 
> I guess it depends on the nature of the crash, it is difficult to answer in general.

I am using gdb tool to run the executable, all i got is "[Thread 26224.0x2480 exited with code 0]" when the gazebo crashed.

WillyTuring, 2024-08-13:
> Even if that case, the answer depends on why it crashed. Even if you do not have any error, you can investigate the nature of the crash by attaching a debugger to the process and obtaining a [...]

So here is the debug output:
1.GUI plugin: Loading up GUI plugin
[Thread 5304.0xb24 exited with code 0]
[New Thread 5304.0x34bc]
[image]

traversaro, 2024-08-13:
It does not seems something related to shared symbols.

closed 2024-08-14 by WillyTuring