gazebosim/gazebo-classic #3393: Same errors on all solutions stipulated in gazebo on Window 11 gazebo==11.14.0=*_*

USeebi, 2024-08-14:
I  got the same errors on all solutions stipulated including, 

conda create -n gazebo gazebo==11.14.0=*_9

conda create -n gazebo gazebo==11.14.0=orignameh160ba46_212

conda create -n gazebo gazebo==11.14.0=gzcompatnameh1a9e25f_12
Is _12 a typo?

Any update on a working gazebo classic distribution from conda-forge for Windows 11?  I don't have problem with that in Windows 10, why Win 11 only?
Any hints?

Many thanks!

_Originally posted by @USeebi in https://github.com/gazebosim/gazebo-classic/issues/3379#issuecomment-2286210323_

USeebi, 2024-08-14:
## Environment
Windows 11
conda info
     active environment : K:\Miniconda3\envs\gazebo
    active env location : K:\Miniconda3\envs\gazebo
            shell level : 2
       user config file : C:\Users\useeb\.condarc
 populated config files : C:\Users\useeb\.condarc
          conda version : 24.7.1
    conda-build version : not installed
         python version : 3.11.9.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=skylake
                          __conda=24.7.1=0
                          __win=0=0
       base environment : K:\Miniconda3\py311  (writable)
      conda av data dir : K:\Miniconda3\py311\etc\conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : K:\Miniconda3\py311\pkgs
                          C:\Users\useeb\.conda\pkgs
                          C:\Users\useeb\AppData\Local\conda\conda\pkgs
       envs directories : K:\Miniconda3\py311\envs
                          C:\Users\useeb\.conda\envs
                          C:\Users\useeb\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/24.7.1 requests/2.32.2 CPython/3.11.9 Windows/10 Windows/10.0.22621 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8 aau/0.4.4 c/. s/. e/.
          administrator : False
             netrc file : None
           offline mode : False

## Description
gazebo --verbose with gzerver.exe & gzclient.exe failed in Application Error

Faulting application name: gzserver.exe, version: 0.0.0.0, time stamp: 0x6615369f
Faulting module name: VCRUNTIME140.dll, version: 14.40.33810.0, time stamp: 0x119a1ae6
Exception code: 0xc0000005
Fault offset: 0x0000000000011288
Faulting process id: 0x0x1FB0
Faulting application start time: 0x0x1DAED8144E4D486
Faulting application path: K:\Miniconda3\envs\gazebo\Library\bin\gzserver.exe
Faulting module path: K:\Miniconda3\envs\gazebo\Library\bin\VCRUNTIME140.dll
Report Id: c3d15bc8-2aff-4417-9024-c3776ed8fc09
Faulting package full name: 
Faulting package-relative application ID: 

Faulting application name: gzclient.exe, version: 0.0.0.0, time stamp: 0x6615369a
Faulting module name: VCRUNTIME140.dll, version: 14.40.33810.0, time stamp: 0x119a1ae6
Exception code: 0xc0000005
Fault offset: 0x0000000000011288
Faulting process id: 0x0x3B10
Faulting application start time: 0x0x1DAED8144EEBECE
Faulting application path: K:\Miniconda3\envs\gazebo\Library\bin\gzclient.exe
Faulting module path: K:\Miniconda3\envs\gazebo\Library\bin\VCRUNTIME140.dll
Report Id: cd3d9729-5cd7-4a76-86ff-ef21d294b933
Faulting package full name: 

## Steps to reproduce
conda create -n gazebo gazebo

traversaro, 2024-08-14:
Which error are you experiencing?

USeebi, 2024-08-14:
> Which error are you experiencing?

gazebo --verbose with gzerver.exe & gzclient.exe failed in Application Error

Faulting application name: gzserver.exe, version: 0.0.0.0, time stamp: 0x6615369f
Faulting module name: VCRUNTIME140.dll, version: 14.40.33810.0, time stamp: 0x119a1ae6
Exception code: 0xc0000005
Fault offset: 0x0000000000011288
Faulting process id: 0x0x1FB0
Faulting application start time: 0x0x1DAED8144E4D486
Faulting application path: K:\Miniconda3\envs\gazebo\Library\bin\gzserver.exe
Faulting module path: K:\Miniconda3\envs\gazebo\Library\bin\VCRUNTIME140.dll
Report Id: c3d15bc8-2aff-4417-9024-c3776ed8fc09
Faulting package full name:
Faulting package-relative application ID:

Faulting application name: gzclient.exe, version: 0.0.0.0, time stamp: 0x6615369a
Faulting module name: VCRUNTIME140.dll, version: 14.40.33810.0, time stamp: 0x119a1ae6
Exception code: 0xc0000005
Fault offset: 0x0000000000011288
Faulting process id: 0x0x3B10
Faulting application start time: 0x0x1DAED8144EEBECE
Faulting application path: K:\Miniconda3\envs\gazebo\Library\bin\gzclient.exe
Faulting module path: K:\Miniconda3\envs\gazebo\Library\bin\VCRUNTIME140.dll
Report Id: cd3d9729-5cd7-4a76-86ff-ef21d294b933
Faulting package full name:

traversaro, 2024-08-14:
Can you report the `conda list` of your environment?

traversaro, 2024-08-14:
By the way, I just tried on my Windows 11 machine and Gazebo starts correctly with this conda list:

~~~
(gazebo) C:\Users\straversaro>conda list
# packages in environment at D:\miniforge3\envs\gazebo:
#
# Name                    Version                   Build  Channel
aom                       3.9.1                he0c23c2_0    conda-forge
assimp                    5.4.1                h0dbab56_0    conda-forge
aws-c-auth                0.7.25               he6981b3_4    conda-forge
aws-c-cal                 0.7.2                h0240d57_1    conda-forge
aws-c-common              0.9.25               h2466b09_0    conda-forge
aws-c-compression         0.2.18               h0240d57_8    conda-forge
aws-c-event-stream        0.4.2               h330dd76_20    conda-forge
aws-c-http                0.8.7                h065a16f_2    conda-forge
aws-c-io                  0.14.18              h3095dc9_2    conda-forge
aws-c-mqtt                0.10.4              h1efefa3_16    conda-forge
aws-c-s3                  0.6.4                h9116173_6    conda-forge
aws-c-sdkutils            0.1.19               h0240d57_0    conda-forge
aws-checksums             0.1.18               h0240d57_8    conda-forge
aws-crt-cpp               0.27.5               h55e8508_5    conda-forge
aws-sdk-cpp               1.11.379             h2205d66_2    conda-forge
azure-core-cpp            1.13.0               haf5610f_0    conda-forge
azure-identity-cpp        1.8.0                h148e6f0_2    conda-forge
azure-storage-blobs-cpp   12.12.0              hf03c1c4_0    conda-forge
azure-storage-common-cpp  12.7.0               h148e6f0_1    conda-forge
blosc                     1.21.6               h85f69ea_0    conda-forge
bullet-cpp                3.25                 h2ab9e98_2    conda-forge
bzip2                     1.0.8                h2466b09_7    conda-forge
c-ares                    1.33.0               h2466b09_0    conda-forge
ca-certificates           2024.7.4             h56e8100_0    conda-forge
cairo                     1.18.0               h91e5215_2    conda-forge
cfitsio                   4.4.1                hc2ea260_0    conda-forge
collada-dom               2.5.0                hc0d5731_8    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
cppzmq                    4.10.0               h42135b4_1    conda-forge
dartsim                   6.14.4               h9059330_2    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.4.1                h63175ca_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
expat                     2.6.2                h63175ca_0    conda-forge
fcl                       0.7.0                he22821c_4    conda-forge
ffmpeg                    7.0.2           gpl_h97ca3ef_100    conda-forge
flann                     1.9.2                h23e6bae_0    conda-forge
fmt                       11.0.2               h7f575de_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_2    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeglut                  3.2.2                he0c23c2_3    conda-forge
freeimage                 3.18.0              h2b56e36_20    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
freexl                    2.0.0                h8276f4a_0    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
gazebo                    11.14.0         orignameh67c379f_216    conda-forge
geos                      3.12.2               h5a68840_1    conda-forge
geotiff                   1.7.3                h232476a_2    conda-forge
getopt-win32              0.1                  hcfcfb64_1    conda-forge
glfw                      3.4                  hcfcfb64_0    conda-forge
glib                      2.80.3               h7025463_2    conda-forge
glib-tools                2.80.3               h4394cf3_2    conda-forge
graphite2                 1.3.13            h63175ca_1003    conda-forge
graphviz                  12.0.0               hb01754f_0    conda-forge
gst-plugins-base          1.24.6               hb0a98b8_0    conda-forge
gstreamer                 1.24.6               h5006eae_0    conda-forge
gts                       0.7.6                h6b5321d_4    conda-forge
[... 153 lines cut ...]
vc                        14.3                h8a93ad2_20    conda-forge
vc14_runtime              14.40.33810         ha82c5b3_20    conda-forge
vs2015_runtime            14.40.33810         h3bf8584_20    conda-forge
wheel                     0.44.0             pyhd8ed1ab_0    conda-forge
x264                      1!164.3095           h8ffe710_2    conda-forge
x265                      3.5                  h2d74725_3    conda-forge
xerces-c                  3.2.5                he0c23c2_1    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
[... cut ...]
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
zeromq                    4.3.5                he1f189c_4    conda-forge
zlib                      1.3.1                h2466b09_1    conda-forge
zstd                      1.5.6                h0ea2cb4_0    conda-forge
zziplib                   0.13.69              h3ca93ac_2    conda-forge
~~~

USeebi, 2024-08-15:
The faulty conda list

```(gazebo) K:\Miniconda3>conda list
# packages in environment at K:\Miniconda3\envs\gazebo:
#
# Name                    Version                   Build  Channel
aom                       3.9.1                he0c23c2_0    conda-forge
assimp                    5.3.1                h0dbab56_3    conda-forge
aws-c-auth                0.7.22              h8c86ca4_10    conda-forge
aws-c-cal                 0.7.1                hea5f451_1    conda-forge
aws-c-common              0.9.23               h2466b09_0    conda-forge
aws-c-compression         0.2.18               hea5f451_7    conda-forge
aws-c-event-stream        0.4.2               h4b8288a_15    conda-forge
aws-c-http                0.8.2                h269d64e_6    conda-forge
aws-c-io                  0.14.10              hfca834b_1    conda-forge
aws-c-mqtt                0.10.4               h519d897_8    conda-forge
aws-c-s3                  0.6.0                hb746b11_2    conda-forge
aws-c-sdkutils            0.1.16               hea5f451_3    conda-forge
aws-checksums             0.1.18               hea5f451_7    conda-forge
aws-crt-cpp               0.27.3               h8c89294_2    conda-forge
aws-sdk-cpp               1.11.329             he0aa860_9    conda-forge
azure-core-cpp            1.13.0               haf5610f_0    conda-forge
azure-identity-cpp        1.8.0                h148e6f0_2    conda-forge
azure-storage-blobs-cpp   12.12.0              hf03c1c4_0    conda-forge
azure-storage-common-cpp  12.7.0               h148e6f0_1    conda-forge
blosc                     1.21.6               h85f69ea_0    conda-forge
bullet-cpp                3.25                 h2ab9e98_2    conda-forge
bzip2                     1.0.8                h2466b09_7    conda-forge
c-ares                    1.33.0               h2466b09_0    conda-forge
ca-certificates           2024.7.4             h56e8100_0    conda-forge
cairo                     1.18.0               h91e5215_2    conda-forge
cfitsio                   4.4.1                hc2ea260_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
cppzmq                    4.10.0               h42135b4_1    conda-forge
dartsim                   6.13.2               h15b4ae7_2    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.4.1                h63175ca_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
expat                     2.6.2                h63175ca_0    conda-forge
fcl                       0.7.0                he22821c_4    conda-forge
ffmpeg                    6.1.2           gpl_h97ca3ef_100    conda-forge
flann                     1.9.2                h23e6bae_0    conda-forge
fmt                       10.2.1               h181d51b_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_2    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeglut                  3.2.2                he0c23c2_3    conda-forge
freeimage                 3.18.0              h2b56e36_20    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
freexl                    2.0.0                h8276f4a_0    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
gazebo                    11.14.0         orignameh160ba46_212    conda-forge
geos                      3.12.2               h5a68840_1    conda-forge
geotiff                   1.7.3                h232476a_2    conda-forge
getopt-win32              0.1                  hcfcfb64_1    conda-forge
glib                      2.80.3               h7025463_2    conda-forge
glib-tools                2.80.3               h4394cf3_2    conda-forge
graphite2                 1.3.13            h63175ca_1003    conda-forge
graphviz                  9.0.0                h51cb2cd_1    conda-forge
gst-plugins-base          1.24.6               hb0a98b8_0    conda-forge
gstreamer                 1.24.6               h5006eae_0    conda-forge
gts                       0.7.6                h6b5321d_4    conda-forge
harfbuzz                  9.0.0                h81778c3_0    conda-forge
hdf4                      4.2.15               h5557f11_7    conda-forge
hdf5                      1.14.3          nompi_h2b43c12_105    conda-forge
[... 128 lines cut ...]
uriparser                 0.9.8                h5a68840_0    conda-forge
vc                        14.3                h8a93ad2_20    conda-forge
vc14_runtime              14.40.33810         ha82c5b3_20    conda-forge
vs2015_runtime            14.40.33810         h3bf8584_20    conda-forge
wheel                     0.44.0             pyhd8ed1ab_0    conda-forge
x264                      1!164.3095           h8ffe710_2    conda-forge
x265                      3.5                  h2d74725_3    conda-forge
xerces-c                  3.2.5                he0c23c2_1    conda-forge
[... cut ...]
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
zeromq                    4.3.5                he1f189c_4    conda-forge
zlib                      1.3.1                h2466b09_1    conda-forge
zstd                      1.5.6                h0ea2cb4_0    conda-forge
zziplib                   0.13.69              h3ca93ac_2    conda-forge```

USeebi, 2024-08-15:
I did update

  added / updated specs:
    - gazebo==11.14.0=orignameh67c379f_216

but it didn't help.  I don't have a second Win 11 system to try at the moment.  Is there any possibility the problem is outside the conda env?

Thanks a lot!