gazebosim/gazebo-classic #3396: Removing and inserting a new model failed

WillyTuring, 2024-08-17:
Hi guys,
In the code, i am trying to remove an old model loaded originally from sdf file, then insert a new one with method "world->InsertModelSDF(*updatedSDF);"; the old one is being removed successfully, but i got error when inserting the new one:
[image]

Any thoghts on the problem? Thanks in advance.

WillyTuring, 2024-08-17:
My hunch is that the old model wasn't being removed cleanly thus we have this problem, we are using "world->RemoveModel(sphereModel);" to remove the model, it did get removed, but i think the resources related didn't get cleaned up completely.

traversaro, 2024-08-19:
Which version of ogre are you using? If I recall correctly, at some point ogre changed the logic related to resources. You can find more information in https://github.com/conda-forge/gazebo-feedstock/issues/74, in particular https://github.com/conda-forge/gazebo-feedstock/issues/74#issuecomment-835388216 . At the conda-forge level, we are using a patch in https://github.com/conda-forge/gazebo-feedstock/pull/75/files that we never ported upstream as it creates a leak in version of ogre <= 1.9.

WillyTuring, 2024-08-19:
> Which version of ogre are you using? If I recall correctly, at some point ogre changed the logic related to resources. You can find more information in [conda-forge/gazebo-feedstock#74](https://github.com/conda-forge/gazebo-feedstock/issues/74), in particular [conda-forge/gazebo-feedstock#74 (comment)](https://github.com/conda-forge/gazebo-feedstock/issues/74#issuecomment-835388216) . At the conda-forge level, we are using a patch in https://github.com/conda-forge/gazebo-feedstock/pull/75/files that we never ported upstream as it creates a leak in version of ogre <= 1.9.

I am using "Ogre Version: 1.12.9".