gazebosim/gazebo-classic #3409: Gazebo 11.15 cannot successfully load the model when simulating with PX4.

MilkyWay2001, 2024-11-30:
## Environment
* OS Version: 
ubuntu 20.04
ros-neotic
gazebo 11.15
px4 15.3

## Description
When I run the PX4 provided script as usual, the Gazebo window opens but shows an error saying it cannot load the model. However, it works fine in Gazebo 11.14 version.

## Steps to reproduce
1. roslaunch px4 posix_sitl.launch

## Output
[image]

[image]

traversaro, 2024-11-30:
> When I run the PX4 provided script as usual 

Can you provide more details on this? As you may imagine we may not know what you "usual script" is.

MilkyWay2001, 2024-11-30:
> > When I run the PX4 provided script as usual
> 
> Can you provide more details on this? As you may imagine we may not know what you "usual script" is.

Okay, I'll paste them below, thank you.

posix_sitl.launch：
```
<?xml version="1.0"?>
<launch>
    <!-- Posix SITL environment launch script -->
    <!-- launches PX4 SITL, Gazebo environment, and spawns vehicle -->
    <!-- vehicle pose -->
    <arg name="x" default="0"/>
    <arg name="y" default="0"/>
    <arg name="z" default="0"/>
    <arg name="R" default="0"/>
    <arg name="P" default="0"/>
    <arg name="Y" default="0"/>
    <!-- vehicle model and world -->
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/$(arg vehicle)/$(arg vehicle).sdf"/>
    <env name="PX4_SIM_MODEL" value="gazebo-classic_$(arg vehicle)" />

    <!-- gazebo configs -->
    <arg name="gui" default="true"/>
    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>
    <arg name="respawn_gazebo" default="false"/>
    <!-- PX4 configs -->
    <arg name="interactive" default="true"/>
    <!-- PX4 SITL -->
    <arg unless="$(arg interactive)" name="px4_command_arg1" value="-d"/>
    <arg     if="$(arg interactive)" name="px4_command_arg1" value=""/>
    <node name="sitl" pkg="px4" type="px4" output="screen"
        args="$(find px4)/build/px4_sitl_default/etc -s etc/init.d-posix/rcS $(arg px4_command_arg1)" required="true"/>

    <!-- Gazebo sim -->
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)"/>
        <arg name="debug" value="$(arg debug)"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="$(arg paused)"/>
        <arg name="respawn_gazebo" value="$(arg respawn_gazebo)"/>
    </include>
    <!-- gazebo model -->
    <node name="$(anon vehicle_spawn)" pkg="gazebo_ros" type="spawn_model" output="screen" args="-sdf -file $(arg sdf) -model $(arg vehicle) -x $(arg x) -y $(arg y) -z $(arg z) -R $(arg R) -P $(arg P) -Y $(arg Y)"/>
</launch>

```
LOG：
```
[rospy.client][INFO] 2024-11-30 06:03:04,469: init_node, name[/vehicle_spawn_ubuntu_2321_8049803062046353785], pid[2451]
[xmlrpc][INFO] 2024-11-30 06:03:04,470: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2024-11-30 06:03:04,471: Started XML-RPC server [http://ubuntu:43971/]
[rospy.impl.masterslave][INFO] 2024-11-30 06:03:04,471: _ready: http://ubuntu:43971/
[rospy.init][INFO] 2024-11-30 06:03:04,472: ROS Slave URI: [http://ubuntu:43971/]
[rospy.registration][INFO] 2024-11-30 06:03:04,474: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2024-11-30 06:03:04,475: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2024-11-30 06:03:04,572: registered with master
[rospy.rosout][INFO] 2024-11-30 06:03:04,573: initializing /rosout core topic
[rospy.rosout][INFO] 2024-11-30 06:03:04,581: connected to core topic /rosout
[rospy.simtime][INFO] 2024-11-30 06:03:04,583: initializing /clock core topic
[rospy.simtime][INFO] 2024-11-30 06:03:04,585: connected to core topic /clock
[rosout][INFO] 2024-11-30 06:03:04,589: Loading model XML from file /home/milkyway/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/models/plane/plane.sdf
[rosout][INFO] 2024-11-30 06:03:04,601: Waiting for service /gazebo/spawn_sdf_model
[rospy.internal][INFO] 2024-11-30 06:03:04,769: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2024-11-30 06:03:07,682: topic[/clock] adding connection to [http://ubuntu:46227/], count 0
[rosout][INFO] 2024-11-30 06:03:07,929: Calling service /gazebo/spawn_sdf_model
[rosout][INFO] 2024-11-30 06:03:18,030: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name plane
[rosout][ERROR] 2024-11-30 06:03:18,034: Spawn service failed. Exiting.
[rospy.core][INFO] 2024-11-30 06:03:18,037: signal_shutdown [atexit]
[rospy.internal][INFO] 2024-11-30 06:03:18,040: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2024-11-30 06:03:18,040: topic[/clock] removing connection to http://ubuntu:46227/
[rospy.impl.masterslave][INFO] 2024-11-30 06:03:18,041: atexit
```

hitsz-qfy, 2024-12-03:
same problem

closed 2024-12-03 by j-rivero