gazebosim/gazebo-classic #3411: Update Broke Spawn Model---Cannot run simulation for class

leegs, 2024-12-02:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
This is the binary build of gazebo11_11.15.0-1~focal.  

## Description

Gazebo is being used in a college course at CWRU and students cannot demonstrate code that works with Gazebo 11.14 if they have updated 11.15.  Please rollback functionality to 11.14 as quickly as possible.  

When spawning models after the 11.15 update, models do not spawn.  The spawning of the robot models into Gazebo is hanging and not returning.

## Steps to reproduce

The specific steps to reproduce this requires using a package set of packages prepared for the course.  It is likely that spawning other models has the same effect, but cannot determine right now.  Installing the ros-noetic-ecse-373-ariac package from this repository and running roslaunch ecse-373-ariac ecse-373-ariac.launch should reproduce the error.

Instructions for the course repository are here:
     https://cwru-ecse-373.github.io/

## Output

The spawn_model node hangs and the model does not appear in Gazebo.

traversaro, 2024-12-02:
Probably this is related to https://github.com/gazebosim/gazebo-classic/issues/3409 .

traversaro, 2024-12-02:
Are you able to replicate the problem without using ROS functionality? If not, it probably could make sense to also open an issue in https://github.com/gazebosim/gazebo-classic/issues/3409 ?

In the meanwhile, you may want to pin the gazebo version you are using to 11.14.0 if that version is working for you. I am not familiar how to do that with apt, but I guess it should be possible: https://help.ubuntu.com/community/PinningHowto .

traversaro, 2024-12-02:
I wonder if this may be related to https://github.com/gazebosim/gazebo-classic/pull/3374 ? fyi @scpeters

traversaro, 2024-12-02:
> In the meanwhile, you may want to pin the gazebo version you are using to 11.14.0 if that version is working for you. I am not familiar how to do that with apt, but I guess it should be possible: https://help.ubuntu.com/community/PinningHowto .

Probably @j-rivero may know more about this.

MilkyWay2001, 2024-12-02:
Hello, have you solved it? I have the same problem.

azeey, 2024-12-03:
I'm not sure why, but building gazebo_ros_pkgs from source with the new version of Gazebo11 fixes the problem for me.

scpeters, 2024-12-03:
> I'm not sure why, but building gazebo_ros_pkgs from source with the new version of Gazebo11 fixes the problem for me.

maybe an ABI change from https://github.com/gazebosim/gazebo-classic/pull/3283?

vigneshmgz, 2024-12-03:
I'm also facing the same issue

scpeters, 2024-12-03:
> > I'm not sure why, but building gazebo_ros_pkgs from source with the new version of Gazebo11 fixes the problem for me.
> 
> maybe an ABI change from #3283?

testing whether #3283 broke ABI in https://github.com/gazebosim/gazebo-classic/pull/3412

hitsz-qfy, 2024-12-03:
same problem, and build `gazebo-ros-pkgs` from source in noetic branch work

azeey, 2024-12-03:
I can confirm that the regression is caused by #3283 and I think it's an "ABI" issue as @scpeters suggested. However, I'm not sure it's a typical ABI issue since it involves templates. This is what I did to test.
1. Remove any existing versions of Gazebo11
1. Checkout and build version 11.11.0 since that's what's currently available in packages.ros.org and what `gazebo_ros_pkgs` is built against. Install to `/usr`
2. Build `gazebo_ros_pkgs` 
3. Verify that spawning works (this should work).
4. Checkout e65ed2e (commit for #3283) in Gazebo11, build, and install it.
5. **Without** rebuilding `gazebo_ros_pkgs` check if spawning works. This did not work for me.
6. Checkout 9b72949 (11.14.0 or e65ed2e~1) in Gazebo11, build, and install it.
7. **Without** rebuilding `gazebo_ros_pkgs` check if spawning works. This works for me.

Note that rebuilding `gazebo_ros_pkgs` with any of the versions fixes the problem, so I'm convinced it's an ABI problem, albeit a weird, template related one.

I used a modified launch file from https://github.com/gazebosim/gazebo-classic/issues/3409

```xml
<?xml version="1.0"?>
<launch>

    <arg name="verbose" default="true"/>
    <arg name="sdf" />

    <!-- Gazebo sim -->
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="gui" value="false"/>
        <arg name="debug" value="false"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="false"/>
        <arg name="respawn_gazebo" value="false"/>
    </include>
    <!-- gazebo model -->
    <node name="$(anon vehicle_spawn)" pkg="gazebo_ros" type="spawn_model" output="screen" args="-sdf -file $(arg sdf) -model test_model"/>
</launch>
```

And ran it with `roslaunch ./test.launch sdf:=$HOME/ws/gazebo-classic/test/models/box.sdf`. 

My suggestion is to revert #3283, make a release, and then in a follow-up PR, look into fixing the armhf issue. Perhaps we can wrap the whole `GZ_SINGLETON_DECLARE` block in an ifdef that detects if the architecture is armhf.

hitsz-qfy, 2024-12-03:
您好，     邮件已收到。

j-rivero, 2024-12-03:
> In the meanwhile, you may want to pin the gazebo version you are using to 11.14.0 if that version is working for you. I am not familiar how to do that with apt, but I guess it should be possible: https://help.ubuntu.com/community/PinningHowto .

The pinning workaround would be the best solution until we resolve the issue, yes.

> My suggestion is to revert #3283, make a release, and then in a follow-up PR, look into fixing the armhf issue. Perhaps we can wrap the whole `GZ_SINGLETON_DECLARE` block in an ifdef that detects if the architecture is armhf.

I agree. I'm going to merge #3412 and release a 11.15.1 version.

j-rivero, 2024-12-03:
11.15.1 Gazebo should be in the repo. Can anyone please confirm if it is working as expected?

leegs, 2024-12-03:
Working!  Thank you for the quick response!!

closed 2024-12-04 by traversaro