gazebosim/gazebo-classic #3414: Build failure with boost 1.87: boost::asio::io_service has been removed

scpeters, 2024-12-17:
## Environment
* OS Version: macOS
* Source or binary build?
  
  

## Description
* Expected behavior: gazebo should compile on macOS
* Actual behavior: it fails to build since boost 1.87 was merged to homebrew-core

## Steps to reproduce

1. Build from source on macOS
2.
3.

## Output

https://build.osrfoundation.org/job/gazebo-install-one_liner-homebrew-amd64/3371/console

~~~
In file included from /tmp/gazebo11-20241217-86675-upeq8d/gazebo-11.15.0/build/gazebo/transport/transport.hh:5:
/tmp/gazebo11-20241217-86675-upeq8d/gazebo-11.15.0/gazebo/transport/IOManager.hh:45:28: error: no type named 'io_service' in namespace 'boost::asio'
      public: boost::asio::io_service &GetIO();
              ~~~~~~~~~~~~~^
~~~

see also:

* https://github.com/osrf/homebrew-simulation/issues/2916#issuecomment-2549141346
* https://github.com/boostorg/asio/issues/110

xdire, 2025-01-10:
@azeey @scpeters, any temporary resolution you found to build gazebo11 on macos given this bug?

scpeters, 2025-01-10:
> @azeey @scpeters, any temporary resolution you found to build gazebo11 on macos given this bug?

see https://github.com/osrf/homebrew-simulation/issues/2916#issuecomment-2584596354 and https://github.com/osrf/homebrew-simulation/issues/2916#issuecomment-2584675020 for a temporary solution on macos